open62541/open62541 #5347: Linker errors on MSVS

andy295, 2022-09-21:
## Description
I'm trying to implement my own server starting from the [official tutorial](https://www.open62541.org/doc/1.3/toc.html), but unfortunately I'm running into some issues.

I tried with Visual Studio 2013 and I got the following results

[image]

Then I tried with Visual Studio 2019 and I got the following results

[image]

For both I reported just some rows.

## Background Information / Reproduction Steps

## Project steps
Here below the steps I followed to build in and include the library: 

  1_     I opened the solution on Visual Studio, it was generate by using CMake (this last configuration into the appropriate section).
  2_     After building the solution, I created a new project and added in the open62541.h and open62541.c files.
  3_     I created the first [simple server](https://www.open62541.org/doc/1.3/tutorial_server_firststeps.html#server-configuration-and-plugins).
  4_     Finally, I compiled the project on Visual Studio and I ran into the above errors.

Used CMake options:

I used the CMake GUI, here below the chosen options

[image]

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.2-512
 - [x] Operating system: Windows 10 Enterprise x64

jpfr, 2022-09-21:
Try to add ws2_32 to the libraries that are linked in.

jpfr, 2022-09-22:
You're welcome :-)

closed 2022-09-22 by jpfr