open62541/open62541 #5351: Server unresponsive when trying to monitoring the server object

opcua-tsn-team-kalycito, 2022-09-23:
## Description

## Background Information / Reproduction Steps

In UAExpert, when dragging the server object (ns=0, "i=2253") in the configuration tab in the events view for monitoring purpose, the server is unresposive and not able to monitor for events.
 
[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON ..
```
[issueServerCrashes.txt](https://github.com/open62541/open62541/files/9631154/issueServerCrashes.txt)

The issue looks similar to - #5338 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master(SHA:4a2de90fd3928309b04e9a983651a55343ea0c1f)
 - [x] Other OPC UA SDKs used (client or server): server(tutorial_server_alarms_conditions)
 - [x] Operating system: Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

opcua-tsn-team-kalycito, 2022-09-27:
Hi @jpfr, Hi @andreasebner ,

This issue was seen from the commits of last 3 months, what are the latest changes that will be causing this issue?
Is anyone working on the fix for this issue?

opcua-tsn-team-kalycito, 2022-10-20:
Hi @jpr, Hi @andreasebner ,

The issue was observed from the commit SHA: 11923b8d164dde7a59dc1293340c05e8206d5b61. Upon further analysis we see the issue is introduced in the event loop implementation.
Is there ongoing activity to resolve the same.

jpfr, 2022-10-26:
Reproduced and solved via #5396.
Please validate and close.
The solution involves a quite heavy refactoring of the A&C code.
For 1.3 and prior, disable multithreading (locking) and this goes away.

jpfr, 2022-11-14:
Assumed as resolved.

closed 2022-11-14 by jpfr