open62541/open62541 #5366: Multithreading considerations

dapaulid, 2022-10-05:
## Description

This is a question about multithreading best practices and a corresponding feature request.

Consider these two API functions:
https://github.com/open62541/open62541/blob/2ed3ef625bcc58d8ddae4723229adb18c8ea2e44/include/open62541/client_highlevel_async.h#L525-L529
https://github.com/open62541/open62541/blob/781562ad9c8f056f6a4148e1eb684fd6e921c4f2/include/open62541/client.h#L622-L626

From application point of view, the first function I call to do something meaningful, while calling the second is my duty to keep everything responsive. Since they're not marked with `UA_THREADSAFE`, I'm not allowed to call both functions concurrently from different threads... which is what I would like to do.

Of course I could use a mutex in the application, but then I will run into issues with the `timeout` parameter: It introduces a latency vs. CPU overhead tradeoff (higher values result in delayed method calls and lower values will waste CPU cycles), which is problematic for our use case (see background info).

Is there already any way to solve this issue?

## Suggestion

I think it would be nice to have some way to "cancel" the `UA_Client_run_iterate` call from a different thread. Like this, I could have a dedicated thread that always listens on the network layer, until some API function needs to be called.

My hope is that this could be implemented along the EventLoop work, e.g. by additionally handling "Cancel Events" along with network events. Alternatively, "API Events" could be used that include all parameters of the API call so that the call could be executed in the context of the EventLoop task (active object pattern). Although "more nice", this is probably harder to implement.

Do you see any use for a feature like this? If we align to approach this somehow, I'd be willing to contribute from our [fork](https://github.com/indel-ag/open62541).

## Background Information

We are integrating the stack on a real-time embedded platform, where we have a lot of concurrency and performance is critical.

jpfr, 2022-10-06:
You raise a good point.
We already support interrupt handling in the event loop.
You could use a “nop” interrupt to cancel the eventloop listening.

And yes, Thread safety is a goal also for the client.
Any contributions - also if they only make one function thread-safe - are highly welcome.

dapaulid, 2022-10-07:
Thanks for the hint! I just came across `check_eventloop_interrupt.c`, it looks promising. I'll give it a try.

closed 2026-10-22 by o6-nbeier