open62541/open62541 #5369: Connecting to opc server on Siemens S71500

kobru70, 2022-10-07:
## Description
I want to connect to a Siemens S7 1500 opc server via Username/Password no Security. Lateron I want to connect it via Basic256-Signed and Username/Password.
When I read the issues here, it is said, that there is requeired a private key. What i have is the certificate, which i got with UAExpert. I extracted the public key with the command 
openssl x509 -pubkey -inform der -noout -in ua_cert.der > ua_pubkey.pem
When I use this, like in the excrypted examples, it does not work.
Everywhere it is spoken of using the privat key. I think I don't understand the key thing.
Why do I need the private key, when my application is just a client?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [1.3.3 ] open62541 Version (release number or git tag):
 - [OpenSSL 3.0.5 ] Other OPC UA SDKs used (client or server): 
 - [ Windows Visual Studio 2010] Operating system:
 - [[2022-10-07 05:57:20.119 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-10-07 05:57:20.123 (UTC+0200)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2022-10-07 05:57:20.119 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-10-07 05:57:20.123 (UTC+0200)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2022-10-07 05:57:20.144 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic128Rsa15 with error code BadInvalidArgument
[2022-10-07 05:57:20.145 (UTC+0200)] info/securitypolicy        The basic256 security policy with openssl is added.
[2022-10-07 05:57:20.149 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic256 with error code BadInvalidArgument
[2022-10-07 05:57:20.149 (UTC+0200)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2022-10-07 05:57:20.154 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Basic256Sha256 with error code BadInvalidArgument
[2022-10-07 05:57:20.154 (UTC+0200)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2022-10-07 05:57:20.157 (UTC+0200)] warn/userland      Could not add SecurityPolicy#Aes128Sha256RsaOaep with error code BadInvalidArgument

 ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

neojaw, 2022-10-07:
> Why do I need the private key, when my application is just a client? I thought, the client fetches the certificate, uses the public key to encrypt the data and sends it to the server. The server can now decrypt the data with its private key. Is that wrong?

That is correct so far, but endpoint security mode goes both ways. So the client also needs its own (!) pair of certificate + private key so it can sign its messages and/or the server can encrypt its messages to the client.

There is a script in the repository that you can use to create self-signed certificates, you just have to configure the server to accept them. https://github.com/open62541/open62541/blob/master/tools/certs/create_self-signed.py

kobru70, 2022-10-10:
Thank you neojaw, now I understand.
Taking a self signed certifivate and its provate key (taken from UaExpert) it now works!

closed 2022-10-10 by kobru70