open62541/open62541 #5378: Monitoring a connection between OPC UA Client and OPC UA Server

ThexDragonxKnight, 2022-10-11:
Dear Community,

First of all, a big thank you for the great software that has been and continues to be built here.

We have an OPC UA server on which our OPC UA client can also connect. What we want is to monitor the connection between the server and the client (for connection status, disconnections, etc.). 

Are there any possibilities to do this, specifications on how to do it or is something like this planned?

I know that this request is not an issue so feel free to move it to the corresponding folder or forum.

Thanks in advance,
ThexDragonxKnight

neojaw, 2022-10-12:
You can query the secure channel state, session state and connection status manually at any time with `UA_Client_getState`. If you are only looking for errors, you can use the return value of the `UA_Client_run_iterate` function to check if something went wrong and querying the state makes sense.

If you want to get notified by any state change, you can register a `stateCallback` function in the client config that gets called whenever any one of the above changes. Check out `examples/client_subscription_loop.c` for an example.

hdabis1234, 2024-04-21:
This works in the case where the opc server drops. My question is what happens if an open62541 client is connected to a remote server and the ethernet cable is pulled out. The states mentioned above do not seem to change. We did go through setting connectivityCheckInterval and setting a callback but that was still not reliable. Any suggestions?

closed 2026-10-20 by o6-nbeier