open62541/open62541 #5388: Building lib fpr WEC7: undeclared identifier UA_POLLOUT/__BitScanForward

fraxh22, 2022-10-19:
I have downloaded today from github master branch 1.3. -  Git-Revision: v1.3.3-undefined

According "Building with cmake on Windows" I have downloaded Python 3.1 and cmake.
I want to use OPCUA library on Windows Embedded Compact 7 with ARM processor, Compiler has to be Visual Studio 2008.
So I started cmake with line:

cmake -DUA_ARCHITECTURE=wec7  -DUA_ENABLE_AMALGAMATION=ON  -DUA_ENABLE_DISCOVERY_SEMAPHORE=OFF .. -G "Visual Studio 9 2008"

Then I could build open6251-amalgamation-header and open62541_amalgamation-source without errors.

Now using the two files open62541.h and open62541.c in my project I get error messages, mainly
error C2065: UA_POLLOUT: undeclared identifier
error C3861: _BitScanForward: identifier not found
and some other errors related to missing UA_POLLOUT

Note: Above procedure did work with revision 1.2 two years ago

Any ideas ? Thank you !

matiashr, 2022-12-05:
how could you build using vs2008, the library requires c99, is that really supported by vs2008 ?

fraxh22, 2022-12-07:
Please find attached amalgamated files that could be compiled with VS2000. Please use compile option /TP

Mit freundlichen Grüßen / best regards

i.A. Franz Hofer
Entwicklung - R. & D.
________________________________
[cid:at_logo_2d3bfbc8-d5ed-4bc5-b025-bdee46831ae9.png]

ATOMA Gesellschaft für automatische Waagen GmbH
Traunreuter Str. 2-4 | D-84478 Waldkraiburg | Tel. +49 8638 961-214 | Fax +49 8638 961-111
***@***.******@***.***> | www.multipond.com<http://www.multipond.com>

Follow us for the latest updates
[cid:linkedin-icon_3641f9d1-7b43-4b03-aff8-368f8ad34176.png]<https://www.linkedin.com/company/multipond-w%C3%A4getechnik-gmbh>[cid:youtube-icon_e7d9ba56-adcf-4883-b0a8-3a37f7825c44.png]<https://www.youtube.com/c/multiheadweigher>[cid:facebook-icon_74638c59-44a7-4721-b7a6-8467e1fb3fc2.png]<https://www.facebook.com/MULTIPOND>[cid:instagram-icon_4e463ccf-89c2-4fab-b0e5-68ccc2f4f3de.png]<https://www.instagram.com/atoma_multipond/?hl=de>[cid:xing-icon_fc66fb35-a945-4682-b82c-614bfb962b4f.png]<https://www.xing.com/companies/unternehmensgruppeatoma-multipond>

Registergericht: Traunstein HRB 71 | Geschäftsführer: Dipl.-Ing. (FH), Dipl.-Wirtschaftsing. (FH) Robert Krämer, Wilhelm L. Krämer, Ing. Aart Plomp
________________________________
Die Informationspflicht gem. DS-GVO finden Sie unter: www.multipond.com/datenschutz<https://www.multipond.com/datenschutz-verwaltungsgmbh.html>

E-Mail Disclaimer

Der Inhalt dieser E-Mail ist vertraulich und ausschließlich für den bezeichneten Adressaten bestimmt. Wenn Sie nicht der vorgesehene Adressat
dieser E-Mail oder dessen Vertreter sein sollten, so beachten Sie bitte, dass jede Form der Kenntnisnahme, Veröffentlichung, Vervielfältigung oder
Weitergabe des Inhalts dieser E-Mail unzulässig ist. Wir bitten Sie, sich in diesem Fall mit dem Absender der E-Mail in Verbindung zu setzen.

This e-mail message including any attachments is for the sole use of the intended recipient(s) and may contain privileged or confidential information.
Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender immediately by
reply e-mail and delete the original message and destroy all copies thereof.

matiashr, 2022-12-12:
Dont see any attachment ?

fraxh22, 2022-12-12:
Please try to download from here:

https://we.tl/t-70E19RHRdQ

Mit freundlichen Grüßen / best regards

i.A. Franz Hofer
Entwicklung - R. & D.
________________________________
[cid:at_logo_2d3bfbc8-d5ed-4bc5-b025-bdee46831ae9.png]

ATOMA Gesellschaft für automatische Waagen GmbH
Traunreuter Str. 2-4 | D-84478 Waldkraiburg | Tel. +49 8638 961-214 | Fax +49 8638 961-111
***@***.******@***.***> | www.multipond.com<http://www.multipond.com>

Follow us for the latest updates
[cid:linkedin-icon_3641f9d1-7b43-4b03-aff8-368f8ad34176.png]<https://www.linkedin.com/company/multipond-w%C3%A4getechnik-gmbh>[cid:youtube-icon_e7d9ba56-adcf-4883-b0a8-3a37f7825c44.png]<https://www.youtube.com/c/multiheadweigher>[cid:facebook-icon_74638c59-44a7-4721-b7a6-8467e1fb3fc2.png]<https://www.facebook.com/MULTIPOND>[cid:instagram-icon_4e463ccf-89c2-4fab-b0e5-68ccc2f4f3de.png]<https://www.instagram.com/atoma_multipond/?hl=de>[cid:xing-icon_fc66fb35-a945-4682-b82c-614bfb962b4f.png]<https://www.xing.com/companies/unternehmensgruppeatoma-multipond>

Registergericht: Traunstein HRB 71 | Geschäftsführer: Dipl.-Ing. (FH), Dipl.-Wirtschaftsing. (FH) Robert Krämer, Wilhelm L. Krämer, Ing. Aart Plomp
________________________________
Die Informationspflicht gem. DS-GVO finden Sie unter: www.multipond.com/datenschutz<https://www.multipond.com/datenschutz-verwaltungsgmbh.html>

E-Mail Disclaimer

Der Inhalt dieser E-Mail ist vertraulich und ausschließlich für den bezeichneten Adressaten bestimmt. Wenn Sie nicht der vorgesehene Adressat
dieser E-Mail oder dessen Vertreter sein sollten, so beachten Sie bitte, dass jede Form der Kenntnisnahme, Veröffentlichung, Vervielfältigung oder
Weitergabe des Inhalts dieser E-Mail unzulässig ist. Wir bitten Sie, sich in diesem Fall mit dem Absender der E-Mail in Verbindung zu setzen.

This e-mail message including any attachments is for the sole use of the intended recipient(s) and may contain privileged or confidential information.
Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender immediately by
reply e-mail and delete the original message and destroy all copies thereof.

beffe, 2022-12-23:
I'm having the same problem - upgrading to v1.3 doesn't compile.

I believe the root cause to be [this PR](https://github.com/open62541/open62541/pull/4704) since the `WSAPoll` function is only available since Windows Vista and it doesn't exist on Windows CE.

jpfr, 2023-05-02:
Closed via #5536

commit 0663874d (2023-05-02): fix(core): disable polling for Windows CE (#5536)

Windows CE does not support socket status determination via
WSAPoll, so that CPU-saving measure needs to be disabled.

Fixes #5388
closed 2023-05-10 by jpfr