open62541/open62541 #5392: After anonymous login is disabled, the client cannot connect to the opcua server through the certificate

LXL94, 2022-10-20:
After anonymous login is disabled, the client cannot connect to the opcua server through the certificate

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-10-20:
Which version of the SDK are you using?
Can you provide a log output? Ideally with debug logging enabled.

LXL94, 2022-10-20:
> 您使用的是哪个版本的 SDK？能否提供日志输出？理想情况下，启用调试日志记录。

[2022-10-20 15:31:06.851 (UTC+0800)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-10-20 15:31:06.852 (UTC+0800)] info/securitypolicy	The Basic128Rsa15 security policy with openssl is added.
[2022-10-20 15:31:06.853 (UTC+0800)] info/securitypolicy	The basic256 security policy with openssl is added.
[2022-10-20 15:31:06.854 (UTC+0800)] info/securitypolicy	The basic256sha256 security policy with openssl is added.
[2022-10-20 15:31:06.855 (UTC+0800)] info/securitypolicy	The Aes128Sha256RsaOaep security policy with openssl is added.
[2022-10-20 15:31:06.855 (UTC+0800)] info/userland	UA_ClientConfig_setDefaultEncryption, Good
[2022-10-20 15:31:06.856 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2022-10-20 15:31:06.856 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2022-10-20 15:31:06.856 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2022-10-20 15:31:06.857 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2022-10-20 15:31:06.857 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2022-10-20 15:31:06.918 (UTC+0800)] info/channel	Connection 3 | SecureChannel 20 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-10-20 15:31:06.918 (UTC+0800)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 0: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 1: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 2: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 3: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 4: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting UserTokenPolicy 0 (certificate) in endpoint 5: configuration doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting UserTokenPolicy 0 (certificate) in endpoint 6: configuration doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 7: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Rejecting endpoint 8: security policy doesn't match
[2022-10-20 15:31:06.967 (UTC+0800)] error/client	No suitable UserTokenPolicy found for the possible endpoints
[2022-10-20 15:31:06.967 (UTC+0800)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
[2022-10-20 15:31:06.967 (UTC+0800)] error/userland	Not connected. Retrying to connect in 1 second
[2022-10-20 15:31:07.968 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2022-10-20 15:31:07.968 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2022-10-20 15:31:07.969 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2022-10-20 15:31:07.969 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2022-10-20 15:31:07.969 (UTC+0800)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2022-10-20 15:31:08.067 (UTC+0800)] info/channel	Connection 3 | SecureChannel 21 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-10-20 15:31:08.067 (UTC+0800)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 0: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 1: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 2: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 3: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 4: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting UserTokenPolicy 0 (certificate) in endpoint 5: configuration doesn't match
[... cut ...]
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Rejecting endpoint 8: security policy doesn't match
[2022-10-20 15:31:08.118 (UTC+0800)] error/client	No suitable UserTokenPolicy found for the possible endpoints
[2022-10-20 15:31:08.118 (UTC+0800)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
[2022-10-20 15:31:08.118 (UTC+0800)] error/userland	Not connected. Retrying to connect in 1 second

-----------------------------------------------------
op62541 1.2.2

LXL94, 2022-10-22:
Example of client code：
——————————————————————————————————
   certificate = loadFile("../certs/opcua_client_third_cert.der");
    privateKey = loadFile("../certs/opcua_client_third_key.der");
    trustList[0] = loadFile("../certs/opcua_server_cert.der");

    cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
    cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");

    retVal = UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey, trustList, 1, NULL, 0);
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "UA_ClientConfig_setDefaultEncryption, %s", UA_StatusCode_name(retVal));

    UA_SecurityPolicy_None(cc->securityPolicies, certificate, &cc->logger);

    cc->clientDescription.applicationUri = UA_STRING_ALLOC("urn:open62541.client.third.application");

    cc->userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_X509IDENTITYTOKEN];
    cc->userTokenPolicy.tokenType = UA_USERTOKENTYPE_CERTIFICATE;

    cc->inactivityCallback        = inactivityCallback; /* Set stateCallback */
    cc->connectivityCheckInterval = 2000;               /* Perform a connectivity check every 2 seconds */

    UA_ByteString_clear(&certificate);
    UA_ByteString_clear(&privateKey);
    UA_ByteString_clear(&trustList[0]);
——————————————————————————————————————

LXL94, 2022-10-23:
In the wireshark, I find that the clientNoce and ClientCertificate values are NULL. I want to know how to set them

LXL94, 2022-10-23:
@jpfr

Muddasir-Shakil, 2022-11-07:
@LXL94 do you also get any error on Server side about ApplicationURI fpor the client cert?

LXL94, 2022-11-11:
> @LXL94，您是否在服务器端收到有关应用程序URI fp或客户端证书的任何错误？

No errors were reported

NoelGraf, 2022-12-12:
@LXL94 
Client authentication with certificates was not possible until now. This has changed with this [PR](https://github.com/open62541/open62541/pull/5422)