open62541/open62541 #5403: How to get node id from monitor item

happybruce, 2022-10-25:
Hi, currently I am using monitor item to monitor data change, I want to get node id from monitor item, I try to find related interface but not found, only get below function,
```c
UA_Subscription *
UA_Server_getSubscriptionById(UA_Server *server, UA_UInt32 subscriptionId);

UA_MonitoredItem *
UA_Subscription_getMonitoredItem(UA_Subscription *sub, UA_UInt32 monitoredItemId);
```
I look at structure of UA_MonitoredItem, node id is store in member itemToMonitor, type is UA_ReadValueId. 

Does open62541 have a interface to get node id from monitor item or other ways to get node id? Thank you very much.

jpfr, 2022-10-26:
The methods you mention are an internal API that is unstable.
What is your use case? We can then think of adding a public (stable) API to support that.

If you need to, use `_getMonitoredItem`.
`UA_MonitoredItem` contains the "itemToMonitor" information.

happybruce, 2022-10-27:
@jpfr 

Thank you for your response. My case is this: at server side, I have a variable, its value will change several times before client connect server, after connecting OK, client want to know past changes of this variable. 

I cannot use historical value, so I try to add monitor item to this var at server side, use array to record its changed value; after connection, client will monitor this var too, then server try to get this monitor item from client and send changed values to client.

  What do you mean `_getMonitoredItem`? Could you give a detailed explanation? Thank you very much.

jpfr, 2022-10-27:
You pasted the definition of `UA_Subscription_getMonitoredItem`.
That's what I was referring to.