open62541/open62541 #5419: How to add image variable

Lifbully, 2022-11-02:
## Description
Hi, I am new of open62541. In UaExpert, I find a function named "image viewer" https://documentation.unified-automation.com/uaexpert/1.5.1/html/image.html. It shows that opc ua can add image variable attribute. Is it also possible for open62541 to do that? 
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):open62541 v1.1.6
 - [ ] Other OPC UA SDKs used (client or server): No
 - [ ] Operating system:Visual Studio 2017 win10 SDK 8.1
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Muddasir-Shakil, 2022-11-03:
image file can be treated as any file is treated in OPC UA (bytestrings), so you can technically load the file in the server as bytestring store in OPC UA variable and you can use OPC UA client to read the variable and may be store it in on your host as file may be .png format. 
The image viewer probably works on the file transfer mechanism on UA Expert. So technically you can use open62541 to read the byte stream from server.

o6-nbeier, 2026-10-20:
Thank you for the question.

Set the DataType to ImagePNG (ns0;i=2003), the ValueRank to scalar and the value to a scalar ByteString holding the PNG bytes.

Closing for now.

closed 2026-10-20 by o6-nbeier