open62541/open62541 #5432: Stack corruption in UA_CreateCertificate

Harmageddon, 2022-11-07:
## Description

Inside the method `UA_CreateCertificate` in the `ua_openssl_create_certificate.c` plugin, the method `add_x509V3ext` gives me the following error when executed in Visual Studio 2015 debugger:

```Run-Time Check Failure #2 - Stack around the variable 'ctx' was corrupted.```

The affected line in the plugin is:

https://github.com/open62541/open62541/blob/bdd3d6213cab1661ea08a5bd94f4067a554134ee/plugins/crypto/openssl/ua_openssl_create_certificate.c#L247

The error is thrown when exiting the `add_x509V3ext` method.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -G "Visual Studio 14 2015 Win64" -DCMAKE_INSTALL_PREFIX=..\install -DPython3_EXECUTABLE=...\python3.exe -DUA_ENABLE_ENCRYPTION=OPENSSL ..
```

Log is not very helpful:

```
[07.11.2022 14:30:05::183] [20708::23104] INFO  [UA::Userland]: Trying to create a certificate.
[07.11.2022 14:30:05::214] [20708::23104] UNKNOWN [UA::SecureChannel]: Create Certificate: Generating RSA key. This may take a while.
```

I'm able to reproduce this with the [server_encryption.c](https://github.com/open62541/open62541/blob/bdd3d6213cab1661ea08a5bd94f4067a554134ee/examples/encryption/server_encryption.c) example.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.3 release or current master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Microsoft Windows 10 Pro 21H2 19044.2130
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue
- [x] openSSL version: 3.0.5

jpfr, 2022-11-09:
This method comes from OpenSSL.
Do you install OpenSSL from a packet manager?
It would be great if you could use a debug build of OpenSSL to have a better visibility into the problem.

Harmageddon, 2022-11-09:
I have built OpenSSL from source and can debug through the functions. Maybe the problem is in the calls of `X509_EXTENSION_free` or something like that. But the exception is thrown when exiting the `add_x509V3ext` method, which is part of open62541.

https://github.com/open62541/open62541/blob/bdd3d6213cab1661ea08a5bd94f4067a554134ee/plugins/crypto/openssl/ua_openssl_create_certificate.c#L72-L84