open62541/open62541 #5435: Unhelpful behaviour of UA_ClientConfig_setDefault and UA_Client_new

ukaea-chah, 2022-11-09:
## Description

The documentation for [Client](http://www.open62541.org/doc/master/client.html) says:

> The usual usage is as follows:
> 1. Create a client configuration with default settings as a starting point
> 2. Modifiy the configuration, e.g. modifying the timeout 
> 3. Instantiate a client with it 
> 4. After shutdown of the client, clean up the configuration (free memory)

There are some problems with this:

- UA_ClientConfig_setDefault sets a certificate validation of AcceptAll, and logs a warning about it, _even if_ you subsequently set a different certificate validation.
- If you don't want that spurious warning to appear on stdout, or at all, you would have to construct and set e.g. a null logger _before_ calling UA_ClientConfig_setDefault, then fix it up to a real logger later.
- If you want to set parts of the configuration different from the defaults, in some cases you have to undo / remove existing settings (taking care to call free() and clear() methods where needed).

Also, UA_Client_new() does not create a functional client: some settings such as timeout and localConnectionConfig.recvBufferSize are left as zero, preventing it establishing a connection to a server.  The user is forced to call UA_Client_getConfig and UA_ClientConfig_setDefault themselves, so there is little if any advantage over doing:
```C
 UA_ClientConfig config;
 memset(&config, 0, sizeof(config));
 UA_ClientConfig_setDefault(&config);
 client = UA_Client_newWithConfig(&config);
```
UA_Client_new also duplicates the code to set up a default event loop.

I would have hoped that UA_Client_new would have returned a usable client, to reduce the amount of boilerplate code needed by a newcomer trying out the library, working through tutorials etc.

I would suggest that:

-  UA_ClientConfig_setDefault should set only those fields which are unset (zero);
-  UA_Client_new should call UA_ClientConfig_setDefault rather than doing a half-hearted job itself;
- The recommended sequence for the general case would then be:
  1. Create an empty client configuration (memset to zero)
  2. Set any fields for which the default is not appropriate e.g. a different timeout or logger
  3. Call UA_ClientConfig_setDefault to set usable values for the remainder
  4. Instantiate a client with UA_Client_newWithConfig

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): bdd3d62
 - [x] Operating system: GNU/Linux

ukaea-chah, 2022-11-10:
While trying to prepare a PR, I came across some further complications.

Firstly, it is unsafe to use UA_ClientConfig_setDefault on a local UA_ClientConfig struct before passing it to UA_Client_newWithConfig, unless that struct persists for the lifetime of the client.  This is because UA_ClientConfig_setDefault creates an event loop and a security policy each with a pointer to the logger instance in the local struct.

Secondly, it is not feasible to use a custom (or shared) event loop in conjunction with UA_Client_new.  This is because UA_Client_new creates a posix event loop and passes it in the to UA_Client_newWithConfig, which immediately uses it by registering a housekeeping callback.  So after UA_Client_new returns, it is too late to replace the event loop.

jpfr, 2022-11-16:
>  -  UA_ClientConfig_setDefault should set only those fields which are unset (zero);
    -  UA_Client_new should call UA_ClientConfig_setDefault rather than doing a half-hearted job itself;
    -  The recommended sequence for the general case would then be:
       1.  Create an empty client configuration (memset to zero)
       2.  Set any fields for which the default is not appropriate e.g. a different timeout or logger
       3.  Call UA_ClientConfig_setDefault to set usable values for the remainder
       4.  Instantiate a client with UA_Client_newWithConfig

If we require all user-code to add the memset (because it is no longer done internally), then we could break a lot of existing software that uses the library.
Since we are talking about a breaking change anyway, we could go even further.
This concept of moving a preexisting config into a client is less and less appealing.
Instead we could memset the internal config in the client and require that the config already within the client structure is edited in-situ.

So we would have the following two methods for creating a new client:
- UA_Client_new (a convenience method that sets defaults)
- UA_Client_newUnconfigured (config is memset to zero)
- The method UA_Client_newWithConfig is removed completely
This leaves open the possibility of connecting an unconfigured client by mistake.
But that could also happen today.

> Firstly, it is unsafe to use UA_ClientConfig_setDefault on a local UA_ClientConfig struct before passing it to UA_Client_newWithConfig, unless that struct persists for the lifetime of the client. [...]

With the above proposal the pointer to the logger would never change.

> Secondly, it is not feasible to use a custom (or shared) event loop in conjunction with UA_Client_new. This is because UA_Client_new creates a posix event loop and passes it in the to [...]

For the server we have
- UA_Server_run (which internally uses the following)
- UA_Server_run_startup
- UA_Server_run_iterate
- UA_Server_run_shutdown

There the house-keeping callbacks are only created during _startup.
For the client we can have a similar logic.
Where the housekeeping callbacks are only registered when the client is stopped / disconnected.
That's a tricky point. As the client can try to auto-reconnect.
So disconnected != inactive.

These are some great proposals.
In order to make this efficient we prefer to get changes piece-meal.
We typically review and green-light the public API in a PR before even looking at the implementation code.

ukaea-chah, 2022-11-22:
> If we require all user-code to add the memset (because it is no longer done internally), then we break a lot of existing software

UA_ClientConfig_setDefault tests the values of several fields (logger.log, eventLoop, securityPoliciesSize, sessionLocaleIds, sessionLocaleIdsSize), expecting zero or valid pointers.  Other fields (e.g. securityMode) are not touched, perhaps implicitly assuming zero value.  So any code which leaves these fields uninitialised before calling UA_ClientConfig_setDefault is already broken.

ErikTempelaarVO, 2026-02-19:
I believe this suggestion has been addressed with this PR:
https://github.com/open62541/open62541/pull/5463

and is present in master, first released in 1.4.0