open62541/open62541 #5452: Server crashes when using array variables

RichiePrime, 2022-11-17:
## Description

When using array variables, we get server-crashes once a day. The variables are updated like that:
(elem is an internal structure containing the source-data, elem->size is the array size)

```
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;
attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
UA_UInt32 arrayDims = {elem->size};
attr.arrayDimensions = &arrayDims;
attr.arrayDimensionsSize = 1;

/* Set the value. The array dimensions need to be the same for the value. */
UA_Float *pArray = new UA_Float[elem->size];
for (int i = 0; i < elem->size; i++) pArray[i] = (float)gVal(elem,i);  // copy the data from internal array variable to temporary pArray ..
UA_Variant_setArray(&attr.value, pArray, elem->size, &UA_TYPES[UA_TYPES_FLOAT]); // .. to be used by "setArray"
attr.value.arrayDimensions = &arrayDims;
attr.value.arrayDimensionsSize = 1;
UA_Server_writeValue(server, elem->OPCnodeID, attr.value);
delete [] pArray;

```
## Background Information / Reproduction Steps

When running the system in debug-mode, the crash is caused by a memory-access violation in a "memcpy" call deep inside an internal callback function. The debugger shows valid calling arguments down to the "memcpy" call, the "esi" register happens to be NULL inside the assembly "memcpy" function. Obviously some stack corruption (memcpy reads its arguments from stack).
The variables are updated cyclically (approximately twice a second) by the code above. Crashes only occur, when OPC clients are active.
Perhaps some initialization in the "attr" variable missing?

Operating System: MicroSoft Windows IOT
Compiler: MicroSoft Visual Studio 2010 (same results with Visual Studio 2012)
Application: MFC based C++ application (32bit)
The total application runs several threads in parallel.

jpfr, 2022-11-18:
Which version of the library?
Is multi-threading enabled in the config?
Do you use methods not marked UA_THREADSAFE?

RichiePrime, 2022-11-18:
_Which version of the library?_
Version 1.3.3

_Is multi-threading enabled in the config?_
How to do that?

_Do you use methods not marked UA_THREADSAFE?_
I need to check that. I just found out, that the problem is not only with array variables. I ran a bigger set of variables now for testing and I'll get the same crash after a while. Obviously the number of variables plays a big role.
I'm currently running a test without any updating of variables (like in the first post) but running all other threads and OPC-client as usual.

But is there any OPC communication to the client at all when values are not changed?

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier