open62541/open62541 #5455: UA_WriterGroup and UA_ReaderGroup definitions

OsamaOAyoub, 2022-11-17:
Hello,

I am writing an application that publishes some data from a custom node set based on the real time pubsub examples.

## Description
I am using the example pubsub_nodeset_rt_publisher as a reference and within the example the file ua_pubsub.h is included which contains UA_WriterGroup and UA_ReaderGroup are defined to be used later.

I was able to build and run the example normally within open62541 directory during the compilation of the library with the examples options enabled, but when I try to create my own implementation, the file ua_pubsub.h is for sure not available and I need to include a certain file to define UA_WriterGroup and UA_ReaderGroup datatypes which I cannot find.

Inside ua_pubsub.h it is written that: "The public configuration structs are defined in include/ua_plugin_pubsub.h", but I also can't find this file to be able to run my implementation.

Used CMake options:
 BUILD_SHARED_LIBS                OFF                                          
 CLANG_FORMAT_EXE                 CLANG_FORMAT_EXE-NOTFOUND                    
 CMAKE_BUILD_TYPE                 RelWithDebInfo                               
 CMAKE_INSTALL_PREFIX             /usr/local                                   
 LIB_BPF                          LIB_BPF-NOTFOUND                             
 UA_ARCHITECTURE                  posix                                        
 UA_BUILD_EXAMPLES                ON                                           
 UA_BUILD_TOOLS                   OFF                                          
 UA_BUILD_UNIT_TESTS              OFF                                          
 UA_ENABLE_AMALGAMATION           OFF                                          
 UA_ENABLE_DA                     ON                                           
 UA_ENABLE_DIAGNOSTICS            ON                                           
 UA_ENABLE_DISCOVERY              ON                                           
 UA_ENABLE_ENCRYPTION             OPENSSL                                      
 UA_ENABLE_ENCRYPTION_TPM2        OFF                                          
 UA_ENABLE_HISTORIZING            OFF                                          
 UA_ENABLE_JSON_ENCODING          ON     
 UA_ENABLE_METHODCALLS            ON                                           
 UA_ENABLE_NODESETLOADER          OFF                                          
 UA_ENABLE_PUBSUB                 ON                                           
 UA_ENABLE_PUBSUB_DELTAFRAMES     OFF                                          
 UA_ENABLE_PUBSUB_ETH_UADP        ON                                           
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF                                          
 UA_ENABLE_PUBSUB_INFORMATIONMO   OFF                                          
 UA_ENABLE_SUBSCRIPTIONS          ON                                           
 UA_ENABLE_SUBSCRIPTIONS_EVENTS   ON                                           
 UA_FORCE_WERROR                  ON                                           
 UA_LOGLEVEL                      300                                          
 UA_MULTITHREADING                100                                          
 UA_NAMESPACE_ZERO                FULL     

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Ubuntu with Preempt-RT kernel
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-11-18:
The public API is in
server_pubsub.h

OsamaOAyoub, 2022-11-18:
@jpfr  thanks for the reply, I had this file included earlier, but I am still getting the following error:

```
error: ‘currentWriterGroup’ was not declared in this scope
  447 |     UA_WriterGroup*   currentWriterGroup;
```

OsamaOAyoub, 2022-11-18:
Thanks for your reply

closed 2022-11-18 by OsamaOAyoub