open62541/open62541 #5460: UA_Client_Service_read can only read at most 100 attributes at once

nutslan, 2022-11-21:
# Description

I am using UA_Client_Service_read to read several attributes at once. When the number of nodes to be read is not more than 100, I can get  their node attributes correctly. But when I attempt to read more, for instance, 101 nodes,  the UA_Client_Service_read api gets failed, with responce whose resultsSize is 0 and results is NULL.

```
void client_read_var(UA_Client* client, char* name[]) {
    int arraySize = sizeof(name)/sizeof(char *);

    UA_ReadValueId *itemArray = (UA_ReadValueId *)malloc(arraySize * sizeof(UA_ReadValueId));
    for (int idx = 0; idx < arraySize; ++idx) {
        UA_ReadValueId_init(&itemArray[idx]);
        itemArray[idx].attributeId = UA_ATTRIBUTEID_VALUE;
    }

    for (int idx = 0; idx < arraySize; ++idx) {
        itemArray[idx].nodeId = UA_NODEID_STRING(1, name[idx]);
    }

    UA_ReadRequest request;
    UA_ReadRequest_init(&request);
    request.nodesToRead = &itemArray[0];
    request.nodesToReadSize = arraySize;

    UA_ReadResponse response = UA_Client_Service_read(client, request);

    UA_StatusCode *retStatusArray = (UA_StatusCode *)malloc(arraySize * sizeof(UA_StatusCode));
    UA_StatusCode retval = response.responseHeader.serviceResult;
    if (retval == UA_STATUSCODE_GOOD) {
        if (response.resultsSize == arraySize) {
            for (idx = 0; idx < arraySize; ++idx)
                retStatusArray[idx] = response.results[idx].status;
        }
        else {
            UA_ReadResponse_clear(&response);
            return;
        }
    }

    for (int idx = 0; idx < arraySize; ++idx) {
        if (retStatusArray[idx] == UA_STATUSCODE_GOOD) {
        UA_DataValue res = response.results[idx];
        if (!res.hasValue) { 
            UA_ReadResponse_clear(&response);
            return;
        }

        UA_Variant out;
        memcpy(&out, &res.value, sizeof(UA_Variant));
        UA_Variant_init(&res.value);

        if (out.type == &UA_TYPES[UA_TYPES_DOUBLE]) {
            printf("%f\n", *((UA_Double*)res.value.data));
        }
        else if (out.type == &UA_TYPES[UA_TYPES_INT16]) {
            printf("%d\n", *((UA_Int16*)out.data));
        }
    }
    UA_ReadResponse_clear(&response);
}
```

# Background Information 

Operating System: MicroSoft Windows 10
Compiler: MicroSoft Visual Studio 2019
Develop Language: C

jpfr, 2022-11-23:
Which version of open62541?
What is the remote server?

nutslan, 2022-11-23:
> 

The version of open62541 is v1.3.2.
The  remote server is a PLC(Programming Logic Controller).

o6-nbeier, 2026-10-20:
Thank you for the question.

The client has no 100-item limit. The threshold is most likely a limit on the PLC's server side, so read the service status, discover the server's operation limits and split the read accordingly (your sample also needs fixes in its count and ownership handling).

Closing for now.

closed 2026-10-20 by o6-nbeier