open62541/open62541 #5462: [-Werror,-Wmissing-braces] while packge for Android on Unreal Engine 5

ffreality, 2022-11-22:
## Description

Hi, I am working on an Unreal Engine 5 OPC UA Client plugin based on open62541. Window side is finished and I would like to extend it to also Android.

I can create necessary ".c" and ".h" files on Ubuntu with CMake while selecting POSIX as my target platform.
Also I am integrate them to Unreal Engine but when I package my project, I have these errors.

Is there any suggestion for that ?

```
open62541.h(17401,29): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:     UA_ExpandedNodeId id = {0};
LogPlayLevel:                                                 ^
LogPlayLevel:                                                 {}

open62541.c(5785,51): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel: const UA_ExpandedNodeId UA_EXPANDEDNODEID_NULL = {0};

LogPlayLevel: Warning:
open62541.c(8097,9): warning: No native IEEE 754 format detected. Use slow generic encoding. [-W#pragma-messages]

open62541.c(212823,52): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:     UA_PubSubChannelDataUDPMC defaultValues = {0, {0}, 0, {0}, 255, true, true, true};
LogPlayLevel:                                                                                          ^
LogPlayLevel:                                                                                          {}

open62541.c(212823,60): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:     UA_PubSubChannelDataUDPMC defaultValues = {0, {0}, 0, {0}, 255, true, true, true};
LogPlayLevel:                                                                                                    ^
LogPlayLevel:                                                                                                    {}

open62541.c(213202,37): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:         struct ipv6_mreq groupV6 = {0};
LogPlayLevel:                                                        ^
LogPlayLevel:                                                         {}

open62541.c(213250,35): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:         struct ip_mreq groupV4 = {0};
LogPlayLevel:                                                    ^
LogPlayLevel:                                                     {}

open62541.c(213268,37): error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
LogPlayLevel:         struct ipv6_mreq groupV6 = {0};
LogPlayLevel:                                                        ^
LogPlayLevel:                                                        {}

```

## Background Information / Reproduction Steps
If I disable PubSub options from CMake and use double curly bracet on
UA_ExpandedNodeId id = {0};
UA_ExpandedNodeId id = {{0}};

I can package my project and works on Android. But I don't think it will be a good solution.

I tried both Ninja and GCC Unix Make commands. Also installed clang-format and CMake can find it.

Used CMake options:
`-DUA_ENABLE_JSON_ENCODING:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_DISCOVERY:BOOL="1" -DUA_ENABLE_HISTORIZING:BOOL="1" -DCMAKE_BUILD_TYPE:STRING="Release" -DUA_ENABLE_DIAGNOSTICS:BOOL="1" -DUA_ENABLE_PUBSUB:BOOL="1" -DUA_ENABLE_ENCRYPTION:STRING="OFF" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" -DUA_ENABLE_PUBSUB_DELTAFRAMES:BOOL="1" -DUA_ENABLE_METHODCALLS:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" `

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.3.4 Release
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: .c and .h generated from Ubuntu, Unreal Engine Integration on Windows 11 VS22, Package Project Target Android
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

ffreality, 2022-11-23:
> Which compiler is this? {0} is a standard idiom in C99.

Unreal Engine 5.0.3
Android NDK 21.4.7075529
Visual Studio 2022

I saw same idiom on while I was working on Windows, too but it didn't create a problem. Only Android gives this issue. I think it is about Android NDK.

Normally, developers can disable some -Werrors with flags when they start their project from Android Studio but Unreal not works like that. We don't have an Android Studio project so we can not tell compiler how it should compile its stuffs.

I know that open62541's standart, purpose and this problem is like a special case but this kind of game engine integrations will happen in the future because industrial metaverse / simulation trends. So, I just seek a solution which I can integrate to my case.  For example more modern idioms of these.

```
open62541.c (5785)
const UA_ExpandedNodeId UA_EXPANDEDNODEID_NULL = {0};

open62541.c (212821)
UA_PubSubChannelDataUDPMC defaultValues = { 0, {0}, 0, {0}, 255, true, true, true};

open62541.c (213202 and 213268)
struct ipv6_mreq groupV6 = {0};

open62541 (213250)
struct ip_mreq groupV4 = {0};

open62541.h (17401)
UA_EXPANDEDNODEID_NODEID(UA_NodeId nodeId) {
    UA_ExpandedNodeId id = {0};     // Just this
    id.nodeId = nodeId;
    return id;
}
```

jpfr, 2022-11-26:
Do double curly braces work for you everywhere?

Then we can do a global search/replace and see if another compiler in the CI complains.

ffreality, 2022-11-26:
> Do double curly braces work for you everywhere?
> 
> Then we can do a global search/replace and see if another compiler in the CI complains.

Not all of them requires multiple curly braces. Some of them requires double, some of them triple and some of them quadruplet. 

these ones work when I changed them like this.

```
UA_ExpandedNodeId id = {{0}};
UA_PubSubChannelDataUDPMC defaultValues = { 0, {{{0}}}, 0, {{{0}}}, 255, true, true, true };
struct ipv6_mreq groupV6 = {{{{0}}}};
```
But I couldn't be sure if these are valid solutions or not. So I wanted to ask you about this.

If this okay we have blueprint exposed unreal engine opc ua client plugin for both windows and android.

jpfr, 2022-11-26:
Should be solved via https://github.com/open62541/open62541/pull/5480.

> If this okay we have blueprint exposed unreal engine opc ua client plugin for both windows and android.

Sure. Please that to our Github wiki as well.
Do you track master or the last stable release?
I'm asking because the client got much more reliable with the new EventLoop model we have on master.

ffreality, 2022-11-26:
When I created issue, I used 1.3.4 released version. But I as your suggestion I tried with master.
I can confirmed that conversion from {0} to "memset" solved our problem.

But also I converted these because they were problematic, too

open62541.c(244940,38) and (244999,38)
from = `struct ipv6_mreq groupV6 = { 0 };`
to
```
struct ipv6_mreq groupV6;
memset(&groupV6, 0, sizeof(struct ipv6_mreq));
```

open62541.c(244981,36)
from = `struct ip_mreq groupV4 = { 0 };`
to
```
struct ip_mreq groupV4;
memset(&groupV4, 0, sizeof(struct ip_mreq));
```

closed 2022-11-26 by ffreality

GoetzGoerisch, 2022-12-02:
@jpfr if I'm not mistaken or overlooked things, the last to conversions needs to be added 

> But also I converted these because they were problematic, too
>
> open62541.c(244940,38) and (244999,38) from = `struct ipv6_mreq groupV6 = { 0 };` to
>
> ```
> struct ipv6_mreq groupV6;
> memset(&groupV6, 0, [...]