open62541/open62541 #5465: Upgrade from 1.0 to 1.3.4: how to handle deprected methods

azoitl, 2022-11-23:
As pointed out in the title I'm about to upgrade our application from using open62541 1.0 to 1.3.4. While it is rather straight forward, only two minor API issues, I noticed that many methods that our existing code is using are now deprecated. 

As part of the migration I would like to clean up this problems, but I couldn't find anywhere a documentation about this deprecation and how open62541 handles deprecation. For me the most important information would be what other methods to use. 

The methods in question all end with `_deleteMembers` and come from some generated files (e.g., `UA_String_deleteMembers`, `UA_LocalizedText_deleteMembers`, `UA_NodeId_deleteMembers`). 

What should I use instead? Is this documented somewhere and I missed it?

jpfr, 2022-11-25:
Hey @azoitl ,

We switched the nomenclature from `_deleteMembers` to `_clear`. Both are identical.
_clear is shorter and more correct. Because both also do a zero-memset and not only free memory.

Just do a global search/replace and it should be good.

We kept both _deleteMembers and _clear around for a few major releases.
But now we have deprecated the old _deleteMembers as having both can leads to confusion.

azoitl, 2022-11-28:
Thx @jpfr. 

I expected something like this. But as I inherited the open62541 code base on our side I was not sure. In other open source projects that I use or contribute such deprecation is documented and typically alternatives are presented. Are there plans for doing that here as well. Can I help with such?

jpfr, 2022-11-28:
Sure, we do this as well.
Breaking changes are documented in the CHANGELOG file in the main directory.
But we did reset the changelog at some time.

> typically alternatives are presented.

Yes. Just replace _deleteMembers with _clear.

azoitl, 2022-11-28:
> Yes. Just replace _deleteMembers with _clear.
@jpfr I understood that. I only thought it should be in some documentation. For me it is fine now. Thx.

closed 2022-12-03 by jpfr