open62541/open62541 #5485: How to change revised lifetime of 600.00s ?

frameworker2019, 2022-11-28:
I have trouble with the revised lifetime for long running client operations, so I want to increase the setting.
But I don't know which attribute has to be changed accordingly:
```
Connection 6 | SecureChannel 7 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
``` 

I've tried to change **requestedSessionTimeout** in the client config, but there's no effect. It will be set always to 600s.

neojaw, 2022-11-29:
You are probably looking for the `secureChannelLifeTime` in the same client config.

frameworker2019, 2022-11-29:
Setting ```secureChannelLifeTime``` seems not to change the lifetime. This is my client code:

```
UA_StatusCode Client(UA_Client **client, const char *endPoint) {

    *client = UA_Client_new();
    UA_ClientConfig *config = UA_Client_getConfig(*client);
    config->secureChannelLifeTime = 86400 * 1000;
    UA_ClientConfig_setDefault(config);
    UA_StatusCode retval = UA_Client_connect(*client, endPoint);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(*client);
    }

    return retval;
}
```

And this is how the server is configured:
```
  // Creates the open62451 server
    UA_Server *server = UA_Server_new();
    UA_ServerConfig *sCfg = UA_Server_getConfig(server);
    sCfg->maxSessions = 500;
    sCfg->maxSecurityTokenLifetime = 60.0 * 60.0 * 1000.0;
    sCfg->maxSecureChannels = 2000;
```

So 3600s is the maximum token life time which sets the revised lifetime if the client is passing 0 (?)

And this is the output on client side and server when started with the configuration mentioned above:
```
[2022-11-29 10:29:19.899 (UTC+0100)] [33mwarn/userland[0m	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/channel[0m	Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/client[0m	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/client[0m	Selected endpoint 0 in URL opc.tcp://localhost:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/client[0m	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/client[0m	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2022-11-29 10:29:19.903 (UTC+0100)] [32minfo/client[0m	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2022-11-29 10:29:19.904 (UTC+0100)] [32minfo/userland[0m	date is: 29-11-2022 9:29:19.904
```

Server:
```
[2022-11-29 10:29:08.401 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://suedlichts-imac.local:4840/
[2022-11-29 10:29:19.900 (UTC+0100)] info/network	Connection 6 | New connection over TCP from ::1
[2022-11-29 10:29:19.901 (UTC+0100)] info/channel	Connection 6 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2022-11-29 10:29:19.903 (UTC+0100)] info/session	SecureChannel 1 | Session "" | Session created
[2022-11-29 10:29:19.903 (UTC+0100)] info/session	SecureChannel 1 | Session "" | ActivateSession: Session activated
[2022-11-29 10:29:19.905 (UTC+0100)] info/session	SecureChannel 1 | Session "" | Closing the Session
[2022-11-29 10:29:19.905 (UTC+0100)] info/channel	Connection 6 | SecureChannel 1 | CloseSecureChannel
[2022-11-29 10:29:19.905 (UTC+0100)] info/network	Connection 6 | Closed
```

I've take a look into **ua_client_connect.c** to find out how the life time is set:
```
 UA_Float lifetime = (UA_Float)response.securityToken.revisedLifetime / 1000;
    UA_Boolean renew = (client->channel.state == UA_SECURECHANNELSTATE_OPEN);
    if(renew) {
        UA_LOG_INFO_CHANNEL(&client->config.logger, &client->channel, "SecureChannel "
                            "renewed with a revised lifetime of %.2fs", lifetime);
    } else {
        UA_LOG_INFO_CHANNEL(&client->config.logger, &client->channel,
                            "SecureChannel opened with SecurityPolicy %.*s "
                            "and a revised lifetime of %.2fs",
                            (int)client->channel.securityPolicy->policyUri.length,
                            client->channel.securityPolicy->policyUri.data, lifetime);
    }
```
and here
```
   /* Prepare the OpenSecureChannelRequest */
    UA_OpenSecureChannelRequest opnSecRq;
    UA_OpenSecureChannelRequest_init(&opnSecRq);
    opnSecRq.requestHeader.timestamp = UA_DateTime_now();
    opnSecRq.requestHeader.authenticationToken = client->authenticationToken;
    opnSecRq.securityMode = client->channel.securityMode;
    opnSecRq.clientNonce = client->channel.localNonce;
    opnSecRq.requestedLifetime = client->config.secureChannelLifeTime;
    if(renew) {
        opnSecRq.requestType = UA_SECURITYTOKENREQUESTTYPE_RENEW;
        UA_LOG_DEBUG_CHANNEL(&client->config.logger, &client->channel,
                             "Requesting to renew the SecureChannel");
    } else {
        opnSecRq.requestType = UA_SECURITYTOKENREQUESTTYPE_ISSUE;
        UA_LOG_DEBUG_CHANNEL(&client->config.logger, &client->channel,
                             "Requesting to open a SecureChannel");
    }
```
Need someone who can explain the the code and how the client is set its renewal aka revised life time.

On the other hand, **ua_services_securechannel.c** the lifetime  will set to the maximum if the client passes 0.
```
 /* Set the lifetime. Lifetime 0 -> set the maximum possible */
    channel->securityToken.revisedLifetime =
        (request->requestedLifetime > server->config.maxSecurityTokenLifetime) ?
        server->config.maxSecurityTokenLifetime : request->requestedLifetime;
    if(channel->securityToken.revisedLifetime == 0)
        channel->securityToken.revisedLifetime = server->config.maxSecurityTokenLifetime;
```

Indeed ```UA_Client_config``` in **client.h** says very clearly:

```
 /**
     * Advanced Client Configuration
     * ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ */

    UA_UInt32 secureChannelLifeTime; /* Lifetime in ms (then the channel needs
                                        to be renewed) */
  ```

neojaw, 2022-11-29:
`UA_ClientConfig_setDefault()` sets a default value for a lot of things, including the secure channel lifetime. You have to set it to your own value *after* the call.

frameworker2019, 2022-11-30:
Changed he client code to this with no effect. The server and client log still shows the fixed 600s lifetime:
```
UA_StatusCode Client(UA_Client **client, const char *endPoint) {

    *client = UA_Client_new();
    UA_ClientConfig *config = UA_Client_getConfig(*client);
    UA_ClientConfig_setDefault(config);
    config->secureChannelLifeTime = 86400 * 1000;
    UA_StatusCode retval = UA_Client_connect(*client, endPoint);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(*client);
    }

    return retval;
}
```

What can I do else?

waltronix, 2023-04-26:
What kind of trouble do you have exactly?

I currently have some kind of similar issue with loosing the client session.
In my case the server offers a `RevisedLifetime` of 600.000 in the `OpenSecureChannelReponse`, but only a `RevisedSessionTimeout` of 30.000 in the `CreateSessionResponse`.
So if I miss to send a request within `RevisedSessionTimeout` my session is closed.

I will have to subscribe to something or trigger requests in a short enough interval if I understood the [reference](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.6) correctly. 

Maybe that's also an option for you.

fecaltransplant, 2024-02-25:
I'm fumbling with this now.  This appears to be a "that's not a bug; that's a feature" thing.

It appears that on the server side, if I get the UA_ServerConfig* config object, I can set config->maxSecurityTokenLifetime and config->maxSessionTimeout to big numbers, I get a max of 3600 seconds for a session.  

On the client side, if I get the UA_ClientConfig* clientConfig object, I can set clientConfig->secureChannelLifeTime to 3600 seconds maximum for a session as well.

But I want _forever_.

The client.h file explains that the secureChannelLifeTime is in ms **then the channel needs to be renewed**.

So, as far as I can understand, in your own software, on the client side, you need to periodically call the following:

` UA_StatusCode UA_EXPORT UA_THREADSAFE
UA_Client_renewSecureChannel(UA_Client *client);  `

You'll find its definition in client.h under "Asynchronous Services".

fecaltransplant, 2024-02-28:
I noticed if I run `UA_StatusCode UA_EXPORT UA_THREADSAFE UA_Client_renewSecureChannel(UA_Client *client);`  like crazy from the client side,  I will occasionally crash the server with an assertion of line 262 in ua_timer.c regarding the UA_TimerTree.  

The error was:  !UA_TimerTree_ZIP_MIN(&t->tree) || UA_TimerTree_ZIP_MIN(&t->tree)->nextTime > nowMonotonic

ChatGPT said this assertion comes from non-sequential times in the server's list of renewal times for connections.  

As a result, I slowed the renewal request rate from the client and I don't see the crash anymore.  However, I can't find anything on the server side to simply ignore renew requests until a certain amount of time has passed.  This would be a nice way to prevent the server from crashing.

closed 2026-10-20 by o6-nbeier