open62541/open62541 #5486: Why does the subscription occasionally timeout？

ysx1993, 2022-11-29:
I am now using the subscription monitoring function in open62541 in a project. The scenario is as follows: we use a PLC to connect multiple windows computers through a switch. The computer configuration is the same, and the subscription group of open62541 is used in each computer to subscribe multiple same variables.The variables actually change at an interval of 500ms, the publish interval for the subscription group is 100ms, the sample time for the monitor item is 100ms, the queue size for the monitor item is set to 50, and the timeout for UA_Client_run_iterate is set to 100ms, but strangely,What are the possible reasons why only one computer occasionally experiences data timeouts?

The following is an abnormal time log:
[image]

jpfr, 2022-11-29:
Which version of open62541 do you use?
A closer look at a network dump could show the culprit.
Or you enable debug logging on open62541.
Then it is clear if the server is keeping publish responses back or if not enough publish requests are being sent.

it is possible that you overload the plc.
Some are known to only handle a small number of subscriptions at a time.

ysx1993, 2022-11-30:
I am using the 1.2.3 version of open62541. This problem has been bothering me for a long time, and the recurrence probability of this problem is very small. It is an occasional problem, but it will affect the operation of the whole project.I also tried to use wireshark to monitor the communication packets between open62541 client and PLC in this abnormal windows computer. Combined with a local log of open62541 client, I found something abnormal.wireshark received the values of three monitoring items at 09:44:40.182, which were 80,79,86 respectively, but only 80 values were printed at 09.44.40.182 of the open62541 client log.79 and 86 were printed with a delay of about 270ms, which seems to be related to the client, but I did not do additional time-consuming processing in the data change callback function, could you please help me analyze the cause?

This is the thread code where I execute UA_Client_run_iterate:
[image]

This is the packet captured by wireshark:
[image]

This is the log printed by open62541 client:
[image]

closed 2026-10-20 by o6-nbeier