open62541/open62541 #5494: OPC UA 1.05 conformance (pub/sub)

basyskom-meerkoetter, 2022-12-02:
This is an umbrella ticket which tracks issues to be resolved to make the pub/sub implementation in open62541 OPC UA 1.05 conformant.

### Issues
- xxx

basyskom-meerkoetter, 2022-12-02:
@andreasebner can you please add the issues you are already aware off?

andreasebner, 2022-12-09:
Some of these open topics are not directly related to 1.05. but still open in the PubSub-Part.

- Complete support for maxStringLenght filed (currently structs with nun fixed length fields are not supported)
- Support of the sampling configuration (sampling and publish can be controlled independently)
- Support of the complete PubSub-State machine (WIP)
- Support of QoS-Settings (Representation in the configs, support in the network layers)
- AMQP (not that important)
- PubSub Security / SKS (WIP)
- Delta-Frame overhaul (support should be evalulated)
- Update of PubSub-Config with a single method (PubSubConfigType) (update configuration without overwriting)
- Check and Tests for all Transport / Network settings (these are partly specific to the transport mapping)
- PubSub Diagnostics not supported
- PubSub NS0 custom nodeset should be checked and missing items evaluated

pradeipk, 2022-12-21:
Hi @basyskom-meerkoetter, what support is available for metadata discovery in subscriber application ? Is it static or dynamic in nature ?

Muddasir-Shakil, 2023-01-18:
@andreasebner should this issue be marked as todo or whishlist. otherwise it will be lost in issues. pits

basyskom-meerkoetter, 2023-01-27:
@andreasebner does the order of this list imply priority?