open62541/open62541 #5495: ua_architecture_append_to_library(rt) fails on Android

mojca, 2022-12-04:
## Description

I'm not able to build the library for Android due to the `-rt` flag. The compiler reports
```
ld.lld: error: unable to find library -lrt
```

The code says:

```cmake
if(NOT APPLE AND (NOT ${CMAKE_SYSTEM_NAME} MATCHES "OpenBSD"))
  ua_architecture_append_to_library(rt)
endif()
```

but apparently excluding only OpenBSD from the list is not sufficient. Could you perhaps add at least Android to the exclusion list as well?

## Background Information / Reproduction Steps

First extract https://dl.google.com/android/repository/android-ndk-r25b-linux.zip on Ubuntu and then try to build the project.

Used CMake options:

```
export NDK_PATH=$HOME/android/android-ndk-r25b
cmake -G Ninja \
    -DCMAKE_TOOLCHAIN_FILE=${NDK_PATH}/build/cmake/android.toolchain.cmake \
    -DANDROID_ABI=arm64-v8a \
    -DANDROID_NDK=${NDK_PATH} \
    -DANDROID_PLATFORM=android-24 \
    -DCMAKE_ANDROID_NDK=${NDK_PATH} \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
    -DCMAKE_SYSTEM_NAME=Android \
    -DCMAKE_SYSTEM_VERSION=24 \
    /path/to/source
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux, building for Android
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2022-12-05:
Select Posix as target platform on Linux (my case ubuntu) CMake. I am working on Unreal Engine plugin and it works perfectly on Android.

mojca, 2022-12-06:
What do you mean by "select Posix as target platform"?

The relevant code that is causing some problems comes from "if posix" and it starts working for me if I delete `ua_architecture_append_to_library(rt)`:
https://github.com/open62541/open62541/blob/c0298a9a6bfcc20cadaf54af640d82cabadfb64e/arch/posix/CMakeLists.txt#L12-L31

mojca, 2022-12-06:
Should I try to come up with a github action building for Android that tries to reproduce the build failure?

ffreality, 2022-12-06:
I am working on an UE5 client plugin and we use Android NDK for package.

In order to build open62 for Android I build it from Ubuntu and selected target platform as POSIX on CMAKE.
I used Posix because it is similiar with Android and I used Ubuntu because there is no cross compile support.
My branch was 21 november master branch and I didn't got any error about `ua_architecture_append_to_library(rt)`

But I disabled some features from CMake because master branch has additional options which 1.3.3 doesn't have. Maybe I could disabled that RT feature, too.

My only error was this and I got it when I tried to package my project as apk.
https://github.com/open62541/open62541/issues/5462

We solved it and right now we have no problem about Android and packaging.