open62541/open62541 #5510: UA_CreateCertificate(UA_CERTIFICATEFORMAT_DER): Parameters (outPrivateKey, outCertificate) may not be cleared with UA_ByteString_clear()

yusufkrgz, 2022-12-12:
The out parameters outPrivateKey, outCertificate (with the type UA_ByteString) may not be cleared with the usual way UA_ByteString_clear(), if the format is UA_CERTIFICATEFORMAT_DER.

If the format is UA_CERTIFICATEFORMAT_DER, than out parameters created with i2d_PrivateKey and i2d_X509 api's are from OpenSSL.
They are created with OPENSSL_malloc() and must be freed with OPENSSL_free().

Solution:
Buffers must be cloned as UA_ByteString, than the output from i2d_PrivateKey and i2d_X509 must be freed with OPENSSL_free().

keba-estr, 2023-01-30:
There seems to be also another issue with i2d_X509. According to the documentation the out pointer is increased. Maybe an intermediate buffer pointer must be used instead of the direct usage of the UA_ByteString.

(see https://www.mediacollege.com/cgi-bin/man/page.cgi?section=3&topic=d2i_X509):
i2d_X509() encodes the structure pointed to by x into DER format.  If
       out is not NULL is writes the DER encoded data to the buffer at *out,
       and increments it to point after the data just written.

yusufkrgz, 2023-01-30:
`For OpenSSL 0.9.7 and later if *out is NULL memory will be allocated
       for a buffer and the encoded data written to it. In this case *out is
       not incremented and it points to the start of the data just written.`

i think it is not the case by OpenSSL3, which i used.

But there was a memory exception as out parameters with UA_ByteString_clear() freed. I thought it is clear as i saw the code how the out parameters allocated.

Like in the PEM case, intermediate buffer needed to clone it again in the out parameter which allocated with UA_ByteString_new()

keba-estr, 2023-01-31:
@yusufkrgz: Please have a look at the PR #5606, whether the memory exception is gone with these changes.

yusufkrgz, 2023-01-31:
#e8e0fa61 last commit seems to be correct and the smarter than a clone.

if it is the last commit than i can have a look at it today. or would you want to let me know, when all changes done?

keba-estr, 2023-01-31:
@yusufkrgz: If the pipeline does not fail, this is my last commit for this PR.

yusufkrgz, 2023-01-31:
thank you it works.

but i just tested it in v1.3.4 branch with your changes. main branch had linker error with openssl. i have no info yet why i got it.

fyi: the exception without fix was:

> Invalid address specified to RtlValidateHeap( 0000000000500000, 0000000002FC88D0 )

and the call stack:
```
 	ntdll.dll!RtlpBreakPointHeap()	Unknown
 	ntdll.dll!RtlpValidateHeapEntry()	Unknown
 	ntdll.dll!RtlValidateHeap()	Unknown
 	KernelBase.dll!HeapValidate()	Unknown
 	ucrtbased.dll!_CrtIsValidHeapPointer(const void * block=0x0000000002fc8900) Line 1407	C++
 	ucrtbased.dll!free_dbg_nolock(void * const block=0x0000000002fc8900, const int block_use=1) Line 904	C++
 	ucrtbased.dll!_free_dbg(void * block=0x0000000002fc8900, int block_use=1) Line 1030	C++
 	ucrtbased.dll!free(void * block=0x0000000002fc8900) Line 32	C++
	open62541.dll!UA_Array_delete(void * p=0x0000000002fc8900, unsigned __int64 size=875, const UA_DataType * type=0x00007ffbcdf3c270) Line 1882	C
 	open62541.dll!String_clear(UA_String * s=0x00000000039be888, const UA_DataType * _=0x00007ffbcdf3c5d0) Line 165	C
 	open62541.dll!UA_clear(void * p=0x00000000039be888, const UA_DataType * type=0x00007ffbcdf3c5d0) Line 1303	C
 	OpcUaClient.dll!UA_ByteString_clear(UA_String * p=0x00000000039be888) Line 477	C++
```

keba-estr, 2023-01-31:
Resolved with PR #5606, can be closed

closed 2023-01-31 by jpfr