open62541/open62541 #5514: is the library able to connect a KEPServer without any security enabled but with user name and password?

hobbitdev, 2022-12-15:
## Description
Excuse me, it's not clear to me whether the library is able to connect a KEPServer without any security enabled but with user name and password only; below its endpoint configuration.

Apparently the code I'm using should be right ...
```
UA_StatusCode status = UA_STATUSCODE_GOOD;

// New client object
UA_Client* cli = UA_Client_new();

// Client configuration
UA_ClientConfig* clicfg = UA_Client_getConfig(cli);
status = UA_ClientConfig_setDefault(clicfg);

// Security
clicfg->securityMode = UA_MESSAGESECURITYMODE_NONE;
clicfg->securityPolicyUri = UA_STRING_NULL;

// Connection
status = UA_Client_connectUsername(cli, "opc.tcp://192.168.85.105:49320", "rossano", "mypassword");
```

Using my code, I get the following status code after calling `UA_Client_connectUsername`
0x80020000 - An internal error occurred as a result of a programming or configuration error

i'm able to connect the server only using the function `UA_Client_connect`

KEPServer endpoint configuratin
[image]

Using "UA Expert" the connection can be achieved.
[image]

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -S "W:\code\libraries\res\utility\open62541\v1.3.4\src" 
	-B "c:\temp\build-opcua" 
	-D "CMAKE_BUILD_TYPE=Release" 
	-D "BUILD_SHARED_LIBS=OFF" 
	-D "UA_NAMESPACE_ZERO=FULL" 
	-D "UA_ENABLE_SUBSCRIPTIONS=ON" 
	-D "UA_ENABLE_ENCRYPTION=OPENSSL" 
	-D "UA_BUILD_SELFSIGNED_CERTIFICATE=ON" 
	-D "UA_ENABLE_METHODCALLS=ON" 
	-D "UA_ENABLE_NODEMANAGEMENT=ON" 
	-D "UA_LOGLEVEL=300" 
	-G "Visual Studio 17 2022" 
	-A X64
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2022-12-16:
I have the same problem with Prosys OPC UA server. But there are some missing things in your project.
1- You need to create load your key and cert files which you (should) created with create_self-signed.py
2- your ApplicationUri and ApplicationName values in code have to be same with your certificate.
ApplicationUri is like this

urn:Something:Something

3- security policy uri shouldn't NONE. It should be one of these
https://profiles.opcfoundation.org/profile (Security>SecurityPolicy Client Server)

4- clicfg->securityMode shouldn't UA_MESSAGESECURITYMODE_NONE
You should define it with your server configuration. For example I configured my  ProSYS OPC UA Simulation Server as Sign&Encrypt and Certificate only.
In real world scenarios, there are lots of factories which disables Anonymous connection.

after that, opc server will get a request from a certificate and you will get only one time connection error. When you say your opc server to trust that requested certificate, your second and other requests will be accepted.

This is what should I expect but even after these I get BadInternalError, too. So I will create an issue, too.

hobbitdev, 2022-12-16:
Thank you @ffreality for your tip, I'ìll give it a try ...
Excuse me, just a confirmation, it is necessary to use encryption because of KEPSERVER, which needs an encrypted login; isn't it ?

ffreality, 2022-12-16:
> Thank you @ffreality for your tip, I'ìll give it a try ... Excuse me, just a confirmation, it is necessary to use encryption because of KEPSERVER, which needs an encrypted login; isn't it ?

I don't know KEPSERVER's requirements but ProSYS does require encryption to pass username and password. Because sending them as plain text can create security risk.

As I say before, even with my hints I got same error too and I writing my issue.

ffreality, 2022-12-17:
Currently, ProSYS can see my certificate and connection gives BadTimeOut rather than BadInternalError. My AppName and AppUri theory is true. But I created certificate with UA_CreateCertficate() function in c++

jpfr, 2023-01-06:
I just tried this out.
The 1.3 and master branches work with Prosys.
But you make sure to use the full DiscoveryUrl when conneting.
**INCLUDING THE APPLICATION PATH** after the port.

In my case this is `opc.tcp://ILT835.pc.iitb.fhg.de:53530/OPCUA/SimulationServer`.

Just added more logging to make this detectable on 1.3 and master.
Here is connection log when using the examples/client_encryption executable.

```
jpfr@virtualbox:~/software/open62541/build$ ./bin/examples/client_encryption opc.tcp://ILT835.pc.iitb.fhg.de:53530/OPCUA/SimulationServer ./server_cert.der ./server_key.der SimulationServer*.der
[2023-01-06 15:50:23.774 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-01-06 15:50:23.787 (UTC+0100)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2023-01-06 15:50:23.793 (UTC+0100)] info/securitypolicy        The basic256 security policy with openssl is added.
[2023-01-06 15:50:23.799 (UTC+0100)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2023-01-06 15:50:23.805 (UTC+0100)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2023-01-06 15:50:23.809 (UTC+0100)] info/eventloop     Starting the EventLoop
[2023-01-06 15:50:23.811 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-01-06 15:50:23.818 (UTC+0100)] info/network       TCP 4   | New connection to "ILT835.pc.iitb.fhg.de" on port 53530
[2023-01-06 15:50:23.825 (UTC+0100)] info/channel       TCP 4   | SC 373        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-01-06 15:50:23.826 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-01-06 15:50:23.829 (UTC+0100)] info/client        Rejecting UserTokenPolicy 0 (username) in endpoint 0: configuration doesn't match
[2023-01-06 15:50:23.829 (UTC+0100)] info/client        Rejecting UserTokenPolicy 1 (certificate) in endpoint 0: configuration doesn't match
[2023-01-06 15:50:23.829 (UTC+0100)] info/client        Selected endpoint 0 in URL opc.tcp://ILT835.pc.iitb.fhg.de:53530/OPCUA/SimulationServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2023-01-06 15:50:23.829 (UTC+0100)] info/client        Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2023-01-06 15:50:23.830 (UTC+0100)] info/channel       TCP 0   | SC 0  | Closed the SecureChannel
[2023-01-06 15:50:23.830 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-01-06 15:50:23.836 (UTC+0100)] info/securitypolicy        The Aes128Sha256RsaOaep security policy channel with openssl is created.
[2023-01-06 15:50:23.837 (UTC+0100)] info/network       TCP 5   | New connection to "ILT835.pc.iitb.fhg.de" on port 53530
[2023-01-06 15:50:23.837 (UTC+0100)] info/network       TCP 4   | Socket closed
[2023-01-06 15:50:23.876 (UTC+0100)] info/channel       TCP 5   | SC 374        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep and a revised lifetime of 600.00s
[2023-01-06 15:50:23.876 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-01-06 15:50:23.911 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-01-06 15:50:23.915 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-01-06 15:50:23.918 (UTC+0100)] info/userland      date is: 6-1-2023 14:50:56.917

[2023-01-06 15:50:23.923 (UTC+0100)] info/securitypolicy        The Aes128Sha256RsaOaep security policy channel with openssl is deleted.
[2023-01-06 15:50:23.923 (UTC+0100)] info/channel       TCP 0   | SC 0  | Closed the SecureChannel
[2023-01-06 15:50:23.923 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2023-01-06 15:50:23.923 (UTC+0100)] info/network       TCP 5   | Socket closed
[2023-01-06 15:50:24.024 (UTC+0100)] info/eventloop     Stopping the EventLoop
[2023-01-06 15:50:24.024 (UTC+0100)] info/network       UDP     | Shutting down the ConnectionManager
[2023-01-06 15:50:24.024 (UTC+0100)] info/network       TCP     | Shutting down the ConnectionManager
[2023-01-06 15:50:24.024 (UTC+0100)] info/eventloop     The EventLoop has stopped
```

If you cannot get it to work, please enable debug logging and send what you see.
Send an email if you want to stay off the public forum.

Regards

closed 2026-10-20 by o6-nbeier