open62541/open62541 #5523: Sign&Encrypt, Username&Password Connections and Unreal Engine Integration Hints

ffreality, 2022-12-18:
## Description
I am working on Unreal Engine client integration like couple months and we made quite good progress so far. So, we would like to give some hints and inform issues.

**1- MBedTLS Issue**
As I mentioned before on my previous closed issue, default MBedTLS (including 3.x) doesn't support "Subject Alternative Names" as known as SAN. Commercial servers (such as Kepware and ProSYS) require a certificate and encryption if you want to use "username and password" method. Also servers only accept that certificate if your client's applicationName and applicationUri values are same with certificate's necessary parameters. 

```
ClientConfig->clientDescription.applicationName = Issuer's CN and Subject's CN
ClientConfig->clientDescription.applicationUri = Subject Alternative Name's URL
```
As you see you need to define your applicationUri on your SAN. For this reason, you can not use MBedTLS, directly.
**You need to do these modifications for adding SAN support to your MBedTLS**
https://mbed-tls.readthedocs.io/en/latest/kb/development/adding_new_subject_alternative_name_support/

**So, using OpenSSL will be easy.**

**2- Specific thing about Unreal Engine**
Unreal already have OpenSSL 1.1.k. So, if you try to include another OpenSSL version, it creates a clash when you try to package your project. There are similiar cases in Unreal forum, too such as like this
https://forums.unrealengine.com/t/linking-new-version-of-openssl-for-packaged-project/413080

But we are looking a better approach to including or updating newer version of OpenSSL

but good thing is, Unreal's OpenSSL has all fundamental platform's OpenSSL

**3- First Time Connection**
When you try to connect a server with a certificate at first time, server will reject it. But it show that certificate in its "Certificates" section. Before another try, you need to "trust" that certificate. At least ProSYS works like this and you just need to right click rejected certificate and press "Trust" button.

**4- Current State and Issue**
Currently we can package our project with OpenSSL 1.1.1k and ProSYS simulation server can see our connection request and certificate. But we get "BadTimeOut" error. 

**Codes:**
UE5 Client Connection = https://gist.github.com/ffreality/8e080bd31df2389c927e8128e2e7ab10
UE5 OpenSSL Certificate Creation = https://gist.github.com/ffreality/f4ea29b7a35beeff13ee1c3a9c457b39
Certificate Loaders = https://gist.github.com/ffreality/8fd06e2dca984581c44a9cbb7ea08727

We export certificate and private key which created by ```UA_CreateCertificate()``` as Base64 string. Because we would like to support both runtime certificate creation & reading and path based reading. So, we use "FFBase64" suffix for runtime created certificates and if ```Automatic File Load``` system finds this keyword in certificate's name, it uses ```UA_ByteString_fromBase64()```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch from December 16's night.
 - [x] Other OPC UA SDKs used (client or server): ProSYS OPC UA Simulation Server
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

ffreality, 2022-12-20:
Updates
1- Unreal Engine 5.1 updated its OpenSSL from ```1.1.1k to 1.1.1n.``` We integrated it to open62 for both Windows and Android.

2- We think best method to use newer versions of OpenSSL is building Unreal Engine from source. So, we can edit ```Engine/Source/ThirdParty/OpenSSL/OpenSSL.Build.cs``` and copy newer version to OpenSSL folder.  We don't know if OpenSSL 3 and 1 version has function breaking differences or not but we will try but of them.

On the other hand Meta Quest Pro requires Oculus custom branch for augmented reality features. So, we should use source build in any case.

jpfr, 2023-01-06:
See the part of the answer here: https://github.com/open62541/open62541/issues/5514

> So, if you try to include another OpenSSL version, it creates clashes when you try to "package" your project.

There is no need to "package" OpenSSL different from what Unreal has.
If Unreal provides a DLL you can just link against that.

> When you try to connect a server with a certificate at first time, server will reject it. But it show that certificate in its "Certificates" section. Before another try, you need to "trust" that [...]

Yes, most servers are configured that way.
That cannot be changed.

> But we get BadTimeOut error.

I think that is not the original error occuring.
Just made an improvement on master so that "later errors" do not mask the "causal origin error".

> Also we used AsyncTask() because UA_Client_connectUsername() freezes Unreal's GameThread before giving BadTimeOut error.

On master we have a much improved support for the asynchronous client.
So you can connect without blocking and without needing a separate thread.

If AsyncTask works for you, that is obviously fine as well.

closed 2023-01-07 by ffreality