open62541/open62541 #5529: What is TIMESTAMP UNSPECIFIED?

vvpnet, 2022-12-21:
Hello colleagues!

I'm having problems here:

https://github.com/FreeOpcUa/python-opcua/blob/2cfc9bc8f01d18e1b92d3410f7fb328df6950bfa/opcua/ua/uatypes.py#L47

The freeOpcUa client cannot correctly read the history from getHistoryData_service_default
because it has UNSPECIFIED defined as "-1" (1601-01-01 12:00AM UTC).
https://github.com/open62541/open62541/blob/ae74dc0071cc58cf96abeeaf7b9c6c39dd2ed286/plugins/historydata/ua_history_database_default.c#L45

And in open62541 UNSPECIFIED as LLONG_MIN "-9223372036854775808".
So is there a TIMESTAMP UNSPECIFIED?

vvpnet, 2022-12-21:
[official response](https://opcfoundation.org/forum/opc-ua-standard/what-is-timestamp-unspecified/#p4359) 
Does open62541 seem to have a problem ?

jpfr, 2023-01-07:
I wouldn’t say we have a problem.
Right now open62541 supports the full Int64 range for DateTime.
In newer specs the range is limited between the year 1601 and 9999.
If another server or client SDK receives values outside of that range, it can transparently make the cutoff.
The spec says nothing about rejecting values outside the cutoff range.
So we loose nothing by supporting the full Int64 range.

The OPC UA DateTime definition is lifted from the Win32 API.
https://learn.microsoft.com/en-us/windows/win32/api/minwinbase/ns-minwinbase-systemtime
Presumably this comes from the old DCOM heritage.
There, negative timestamps are also possible.

I perceive the cutoff as overhead. And not particularly useful.
Because time stamps are usually used in a DataValue structure.
And therein we have an extra Boolean flag to make undefined timestamps explicit.

If you really need this, feel free to explain your use case and propose an API change.
We should discuss the API changes before dropping into the implementation.

vvpnet, 2023-01-09:
Thank you so much for the answer!

closed 2023-01-09 by vvpnet