open62541/open62541 #5540: Virtual Nodes, Callback/Hook into Browse Service

AlfonsoKarlsson, 2022-12-28:
Is there a way to hook into the browse service, so that when a client browses a specific node, I can tell the client about references to nodes that do not exist yet, or exist only on demand.  

The reason is that, I would like to implement virtual nodes, that are not stored anywhere in the server, but the server creates the illusion for its clients that these nodes exist. For instance I have a fixed mapping from NodeIds to files on the harddrive. There is a huge amount of files so that it is infeasible to mirror all of them in address space. When the client browses some top level node, that represents the root of the filesystem, he gets the references to all (virtual) filenodes, and can then use the NodeIds of the filenodes to access the underlying file.

jpfr, 2023-01-07:
You can implement this by customizing the node store plugin.

Your server can the generate references on the fly when a node is accessed for browsing.

o6-nbeier, 2026-10-20:
Closing for now.

closed 2026-10-20 by o6-nbeier