open62541/open62541 #5541: BadSecurityChecksFailed when WriteRequest or ReadRequest in different threads at same times

caojianyeah, 2022-12-29:
when write request and read request in different threads at same times, or two write request in different threads at same times, it will happend .

log1 :
[2022-12-28 18:59:19.967 (UTC+0800)] debug/channel[0m	Connection 4 | SecureChannel 30525 | Sending request with RequestId 1338 of type WriteRequest
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 207
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Add 12 bytes of padding plus 1 padding size bytes
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of length 240
[2022-12-28 18:59:19.967 (UTC+0800)] debug/channel[0m	Connection 4 | SecureChannel 30525 | Sending request with RequestId 1339 of type WriteRequest
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 225
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Add 10 bytes of padding plus 1 padding size bytes
[2022-12-28 18:59:19.967 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of length 256
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Verifying chunk signature
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Calculated padding size to be 16
[2022-12-28 18:59:19.976 (UTC+0800)] [32minfo/client[0m	Error receiving the response with status code BadSecurityChecksFailed
[2022-12-28 18:59:19.976 (UTC+0800)] [33mwarn/channel[0m	Connection 4 | SecureChannel 30525 | Receiving the response failed with StatusCode BadSecurityChecksFailed
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 59
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Add 0 bytes of padding plus 1 padding size bytes
[2022-12-28 18:59:19.976 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30525 | Send from a symmetric message buffer of length 65536 a message of length 80
[2022-12-28 18:59:19.977 (UTC+0800)] [32minfo/client[0m	Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good

log2:
[2022-12-28 20:28:30.997 (UTC+0800)] debug/channel[0m	Connection 4 | SecureChannel 30533 | Sending request with RequestId 1395 of type ReadRequest
[2022-12-28 20:28:30.997 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 20:28:30.997 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 1011
[2022-12-28 20:28:30.997 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Add 8 bytes of padding plus 1 padding size bytes
[2022-12-28 20:28:30.998 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of length 1040
[2022-12-28 20:28:31.009 (UTC+0800)] debug/channel[0m	Connection 4 | SecureChannel 30533 | Sending request with RequestId 1396 of type WriteRequest
[2022-12-28 20:28:31.009 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 20:28:31.010 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 207
[2022-12-28 20:28:31.010 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Add 12 bytes of padding plus 1 padding size bytes
[2022-12-28 20:28:31.010 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Verifying chunk signature
[2022-12-28 20:28:31.010 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of length 240
[2022-12-28 20:28:31.011 (UTC+0800)] [33mwarn/channel[0m	Connection 4 | SecureChannel 30533 | Could not verify the signature
[2022-12-28 20:28:31.011 (UTC+0800)] [33mwarn/channel[0m	Connection 4 | SecureChannel 30533 | Receiving the response failed with StatusCode BadSecurityChecksFailed
[2022-12-28 20:28:31.011 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Prepare a symmetric message buffer of length 65536 with a usable maximum payload length of 65515
[2022-12-28 20:28:31.011 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of header+payload length of 59
[2022-12-28 20:28:31.011 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Add 0 bytes of padding plus 1 padding size bytes
[2022-12-28 20:28:31.011 (UTC+0800)] trace/channel[0m	Connection 4 | SecureChannel 30533 | Send from a symmetric message buffer of length 65536 a message of length 80
[2022-12-28 20:28:31.012 (UTC+0800)] [31merror/channel[0m	Connection 4 | SecureChannel 30533 | Received an ERR response with StatusCode BadSecurityChecksFailed and the following reason: 
[2022-12-28 20:28:31.012 (UTC+0800)] [33mwarn/channel[0m	Connection 4 | SecureChannel 30533 | Receiving the response failed with StatusCode Good。

jpfr, 2023-01-06:
which version of the library?
is multi-threading activated in the build settings?

LiyqSmile, 2023-04-24:
I have the same problem. The multi_threading is not activated . I want to know how to solve this problem.

caojianyeah, 2023-06-02:
> I have the same problem. The multi_threading is not activated . I want to know how to solve this problem.

add lock when read/write.