open62541/open62541 #5543: Self signed certificates no longer accepted with OpenSSL version 3.0.2

kjeldflarup, 2023-01-02:
## Description

After upgrading my host to Ubuntu 22.04 I also upgraded to libssl 3.0.2  
Now I get this error on the server side:
```
Connection 5 | Processing the message failed with error BadCertificateUntrusted
```
Debugging the error I find that it is openssl which returns X509_V_ERR_DEPTH_ZERO_SELF_SIGNED_CERT

This seems new to me, is it a new feature of openssl. Anyone knows if openssl can be configured to ignore this, or if I need to modify some code to make my server accept self signed certificates.

I know that I can add the certificate to the trusted list, but I also knows that customers will cry by this sudden change.

## Background Information / Reproduction Steps

I recompiled with openssl version 3.0.2,  
I have not yet made a self contained example.

Used CMake options:

```
EXTRA_OECMAKE="-DCMAKE_BUILD_TYPE=RelWithDebInfo"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DUA_NAMESPACE_ZERO=REDUCED -DUA_ENABLE_SUBSCRIPTIONS=true -DUA_MULTITHREADING=150"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DUA_ENABLE_ENCRYPTION=true -DUA_ENABLE_ENCRYPTION_OPENSSL=true"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DUA_BUILD_EXAMPLES=true"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=false -DUA_ENABLE_PUBSUB_DELTAFRAMES=false -DUA_ENABLE_PUBSUB=false"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DUA_LOGLEVEL=100"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DBUILD_SHARED_LIBS=ON"
EXTRA_OECMAKE="$EXTRA_OECMAKE -DJSON_BuildTests=OFF"
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.3.4
 - [ ] Other OPC UA SDKs used (client or server): uaexpert and open62541 used as client
 - [ ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2023-01-03:
I don't think it is an OpenSSL based problem. We use three certificate creation method for our Unreal Engine 5 opc ua client project. First one uses Windows CLI and 3.0.7 binary to create certificates with a configuration file. Other one uses PEM_write_bio_X509 (we tried 3.0.7 and 1.1.1n) and both of them give same error as you on ProSYS.

(we support UA_CreateCertificate() but its definable issuer and SAN tags are insufficient. Also when we export UA_ByteString as Base64, PEM_read_bio_X509 can not read its values.)

On the other hand even we add our certificate to trusted list, we got BadTimeOut error on ProSYS. If you can connect a ProSYS or Kepware server, I would like to know your workflow.

because we paid attention to
applicationUri = SAN -> URI (including format)
applicationName = Issuer -> CN
and we still get timeout error with UA_Client_connectUsername() on Sign&Encrypt and Sign methods.

kjeldflarup, 2023-01-03:
I found that I have to call 
```
UA_CertificateVerification_AcceptAll( &open62541_server_config->certificateVerification );
```

And now this also has to be done in the client.

closed 2023-01-03 by kjeldflarup