open62541/open62541 #5556: How to create an OPC-UA View?

tboeckel, 2023-01-11:
Hi!

## Description
There are lots of examples how to create variables within the Objects folder of an OPC-UA server. But there is no example of how to populate the Views folder? In principe it should just contain references to variables in the Objects folder. Correct?

## Background Information / Reproduction Steps
We have several OPC-UA servers, each with its specific structure in the Objects folder. As usual in software development, things are changing day by day, stuff is added or deleted or moved around for more consistency. Although it is easy for us to change this structure, our customers often cannot replicate these changes themselves, because their software is developed by other facilities and each changes requires lots of time (and involves paying money aswell). They just want to access the same NodeIds to keep their software unchanged, no matter how much had to be changed on our side.

So our idea is to use OPC-UA views to work around this issue. This way we could modify our variable tree in the Objects folder as we require it. The entries in the View folder on the other hand will be kept the same all the time and only the references to the Objects folder are changed. That way a customer will always access a "constant" View, while the Objects folder can be rearranged and modified without any visible changes in the View.

Some examples programs on how to setup a View would be highly appreciated.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux, Windows, any
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-01-17:
We currently don't support views.
It would be cool to have. But frankly we haven't seen a lot of users asking for it.

I added this to the wishlist in #5075.

daniel-schmidt, 2023-06-16:
We use Views with open62541 to organize our whole data model into sub-parts for results with and without errors. There is `UA_Server_addViewNode` which seems to work fine. You can definitively add nodes with this function to the top-level Views folder and they show up correctly in clients like UaExpert.

Unfortunately, you cannot raise Events from View nodes, i.e. in our case to get notified when there was a non-successful result. When you try it, you hit this TODO: https://github.com/open62541/open62541/blob/0b1d72f157c6a8f0c05aca56896b217c816e9827/src/server/ua_subscription_events.c#L302

So official support for Views with Events would be appreciated on my side!

GoetzGoerisch, 2023-06-16:
From my experience _Views_ are basically deprecated, as they do not satisfy the use case they where thought out for.

One can reference Nodes always in other hierarchies.
NodeIds are not to be considered static anyway. Only the Entry point and the browsepath shall stay the same.

Therefore Companion Specifications are available to define Types to support this.