open62541/open62541 #5570: error conversion to UA_UInt16

eziosella, 2023-01-16:
## Description
Hi all and thanks for your wonderful project.
I am trying to compile it without success on a lubuntu 16.04.7 LTS but I get the error below:

## Background Information / Reproduction Steps
theese are my steps:
```bash
git clone https://github.com/open62541/open62541.git
cd open62541/
git submodule update --init --recursive
mkdir build && cd build
cmake -DCMAKE_INSTALL_PREFIX=/opt/open62541 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=Full -DBUILD_SHARED_LIBS=On -DUA_MULTITHREADING=0 ..
make -j
```
what I obtain is:

```bash
[  1%] Built target open62541-generator-statuscode
[  1%] Built target open62541-generator-transport
[  3%] Built target open62541-generator-types
[  5%] Built target open62541-generator-namespace
Scanning dependencies of target open62541-plugins
[  5%] Built target open62541-code-generation
[  7%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_common.c.o
[  9%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_log_stdout.c.o
[  9%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix.c.o
[  9%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_udp.c.o
[ 24%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_eth.c.o
[ 11%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_interrupt.c.o
[ 13%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_log_syslog.c.o
[ 13%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_epoll.c.o
[ 13%] Building C object CMakeFiles/open62541-plugins.dir/arch/posix/ua_clock.c.o
[ 13%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_nodestore_ziptree.c.o
[ 13%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/ua_securitypolicy_none.c.o
[ 15%] Building C object CMakeFiles/open62541-plugins.dir/plugins/crypto/ua_pki_none.c.o
[ 16%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_select.c.o
[ 18%] Building C object CMakeFiles/open62541-plugins.dir/arch/posix/ua_architecture_functions.c.o
[ 18%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_config_default.c.o
[ 20%] Building C object CMakeFiles/open62541-plugins.dir/arch/eventloop_posix_tcp.c.o
[ 22%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_accesscontrol_default.c.o
[ 26%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_nodestore_hashmap.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_async.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_networkmessage.c.o
[ 30%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o
[ 32%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_ns0.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_config.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_utils.c.o
[ 35%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writer.c.o
[ 37%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_dataset.c.o
[ 39%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_ns0_diagnostics.c.o
[ 35%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_readergroup.c.o
[ 39%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_ns0.c.o
[ 39%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_connection.c.o
[ 41%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_discovery.c.o
[ 43%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_reader.c.o
[ 45%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_manager.c.o
[ 43%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_writergroup.c.o
[ 47%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o
[ 49%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o
[ 49%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_securitygroup.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/pubsub/ua_pubsub_keystorage.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o
[ 54%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_method.c.o
[ 52%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o
[ 56%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o
[ 56%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o
[ 56%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_monitoreditem.c.o
[... 55 lines cut ...]
CMakeFiles/open62541-object.dir/build.make:354: set di istruzioni per l'obiettivo "CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o" non riuscito
make[2]: *** [CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o] Errore 1
make[2]: *** Attesa per i processi non terminati....
/home/vx5000/open62541/src/server/ua_subscription_events_filter.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-static-in-inline’ [-Werror]
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:1554: set di istruzioni per l'obiettivo "CMakeFiles/open62541-object.dir/src/server/ua_subscription_events_filter.c.o" non riuscito
make[2]: *** [CMakeFiles/open62541-object.dir/src/server/ua_subscription_events_filter.c.o] Errore 1
```

I have already compile the same commit project with success on a Ubuntu 22.04 64bit, on a Lubuntu 18.04 64bit

Used CMake options:

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.3.4 (commit f4db84ae8b0a267a4a2be6ebe445d9832640e583)
 - [ ] Other OPC UA SDKs used (client or server): none
 - [ ] Operating system: Ubuntu 16.04.7 LTS on Virtualbox
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

janfelixklein, 2023-01-16:
I just tried to build open62541 on a similar system with Ubuntu 16.04 and ran into a similar error. Also no problems with it on e.g. Ubuntu 22.04

eziosella, 2023-01-16:
> I just tried to build open62541 on a similar system with Ubuntu 16.04 and ran into a similar error. Also no problems with it on e.g. Ubuntu 22.04

Thanks janfelixklein for your quickly test and feed-back, at least it means I am not a single case so far.
I am guessing it could be related with the gcc version, in my case I am using "gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609", what about you?

jpfr, 2023-01-16:
Hey @eziosella @janfelixklein,

These look like pretty bening warnings that get amplified by our warning==error compiler flags.
The warnings seem rather specific to GCC5. As we have a couple other GCC running in the CI without that problem.

If you can contribute a bit the solution can be integrated much faster.
As the core has a lot on its plate atm.

Three possible fixes in increasing order of effort:
- Turn down some of the compiler flags on your end.
- Add more explicit integer casting to the code and send us a patch
- Add a legacy Ubuntu build environment to our CI (in the /.github folder) and send the patch

eziosella, 2023-01-17:
> Hey @eziosella @janfelixklein,
> 
> These look like pretty bening warnings that get amplified by our warning==error compiler flags. The warnings seem rather specific to GCC5. As we have a couple other GCC running in the CI without that problem.
> 
> If you can contribute a bit the solution can be integrated much faster. As the core team has a lot on its plate atm.
> 
> Three possible fixes in increasing order of effort:
> 
>     * Turn down some of the compiler flags on your end.
> 
>     * Add more explicit integer casting to the code and send us a patch
> 
>     * Add a legacy Ubuntu build environment to our CI (in the /.github folder) and send the patch

Hey @jpfr,

Thanks for your reply, it is exactly as you suggested.
I have just compiled properly just adding in the cmake parametrization "-DUA_FORCE_WERROR=Off".

As you wrote, this is just one of the possible solutions, probably the easiest way.

I would like to contribute trying the solve with the better way in my opinion, resolving the cast warnings.

For the third solution, I am not sure of what you meant ('cause my poor technical english, sorry).

So, at the end, the quickly solution was:
```bash
cmake -DCMAKE_INSTALL_PREFIX=/opt/open62541 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=Full -DBUILD_SHARED_LIBS=On -DUA_MULTITHREADING=0 -DUA_FORCE_WERROR=Off ..
```

I am not an expert in this kind of staff, should I close the issue, or in case I would like to add an extra solution it will be better keep it opened?

Thanks.

jpfr, 2023-01-17:
I will adjust the title and we can keep it open.
As only very few people are affected and it is "only" a warning I cannot assign this developer resources right away.

But community members might be interested to pick this up and provide a code-level solution that adds more casting.

GoetzGoerisch, 2023-01-17:
@eziosella may I ask why you are using `ubuntu-16.04`?

Using this in the GitHub and Azure CIs is already no longer possible, even `ubuntu-18.04` is already deprecated.
As ubuntu releases in Extended Security Maintenance is not available to the general public.

eziosella, 2023-01-17:
> @eziosella may I ask why you are using `ubuntu-16.04`?
> 
> Using this in the GitHub and Azure CIs is already no longer possible, even `ubuntu-18.04` is already deprecated. As ubuntu releases in Extended Security Maintenance is not available to the general public.

I was just wondering if the project was re-usable in some older machines we have, without be forced to O.S upgrading.

FengJi2021, 2023-01-25:
I have a similar problem while compiling in a emulator (qemu) with pi3
`[ 86%] Building C object CMakeFiles/open62541-object.dir/deps/base64.c.o
[ 86%] Building C object CMakeFiles/open62541-object.dir/deps/itoa.c.o
[ 86%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_subscription_monitoreditem.c.o
[ 88%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_subscription_events_filter.c.o
[ 90%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o
[ 90%] Building C object CMakeFiles/open62541-object.dir/deps/parse_num.c.o
cc: fatal error: Killed signal terminated program cc1
compilation terminated.
make[2]: *** [CMakeFiles/open62541-object.dir/build.make:786: CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o] Error 1
{standard input}: Assembler messages:
{standard input}:7087: Warning: end of file not at end of a line; newline inserted
{standard input}:7956: Error: unknown pseudo-op: `.asc'
make[2]: *** Deleting file 'CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o'
make[2]: *** Waiting for unfinished jobs....
cc: fatal error: Killed signal terminated program cc1
compilation terminated.
make[2]: *** [CMakeFiles/open62541-object.dir/build.make:604: CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o] Error 1
make[2]: *** Deleting file 'CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o'`

Later it will be a kernel bug
`[ 4930.030138] kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:1085!
[ 4930.094347] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[ 4930.108195] Modules linked in: cfg80211 rfkill 8021q garp stp llc cdc_ether cdc_subset raspberrypi_hwmon uio_pdrv_genirq uio ip_tables x_tables ipv6
[ 4930.174808] CPU: 2 PID: 315 Comm: wpa_supplicant Not tainted 5.10.103-v8+ #1529
[ 4930.259473] Hardware name: Raspberry Pi 3 Model B+ (DT)
[ 4930.272350] pstate: 80000085 (Nzcv daIf -PAN -UAO -TCO BTYPE=--)
[ 4930.273420] pc : bcm2835_sdhost_finish_data+0xec/0xf0
[ 4930.273756] lr : bcm2835_sdhost_dma_complete+0x170/0x218
[ 4930.274013] sp : ffffffc0115cbd30
[ 4930.274321] x29: ffffffc0115cbd30 x28: 0000000000000006
[ 4930.274653] x27: ffffffc010d6a000 x26: ffffffc0115cbdf8
[ 4930.274933] x25: ffffffc010e1cbf8 x24: dead000000000100
[ 4930.275141] x23: dead000000000122 x22: ffffff80021cdac0
[ 4930.275393] x21: 0000000000000000 x20: 0000000000000000
[ 4930.275773] x19: ffffff8002ba2540 x18: 0000000000000000
[ 4930.275911] x17: 0000000000000000 x16: 0000000000000000
[ 4930.276028] x15: 0000000000000000 x14: 0000000000000000
[ 4930.276146] x13: 0000000000000000 x12: 0000000000000000
[ 4930.276447] x11: 0000000000000040 x10: ffffffc011315628
[ 4930.276721] x9 : ffffffc0108b8798 x8 : ffffff8020c606c0
[ 4930.276952] x7 : 0000000000000000 x6 : 0000000000001ff4
[ 4930.277160] x5 : 0000000011d68000 x4 : 00000000c0000000
[ 4930.277413] x3 : 0000000000000102 x2 : 0000000000000101
[ 4930.277675] x1 : ffffff80021cdac0 x0 : ffffff8002ba2540
[ 4930.278135] Call trace:
[ 4930.282912]  bcm2835_sdhost_finish_data+0xec/0xf0
[ 4930.283118]  bcm2835_sdhost_dma_complete+0x170/0x218
[ 4930.335842]  vchan_complete+0x1f8/0x248
[ 4930.342677]  tasklet_action_common.isra.20+0x15c/0x1a0
[ 4930.342944]  tasklet_action+0x40/0x50
[ 4930.343328]  __do_softirq+0x1a8/0x510
[ 4930.354469]  irq_exit+0xe8/0x108
[ 4930.366483]  __handle_domain_irq+0xa0/0x110
[ 4930.367266]  bcm2836_arm_irqchip_handle_irq+0x6c/0x80
[ 4930.367600]  el1_irq+0xcc/0x180
[ 4930.367840]  handle_mm_fault+0x350/0xcb8
[ 4930.368148]  do_page_fault+0x148/0x3e8
[ 4930.368881]  do_mem_abort+0x48/0xa8
[ 4930.380020]  el0_ia+0x68/0xc8
[ 4930.394972]  el0_sync_compat_handler+0xdc/0x110
[ 4930.395980]  el0_sync_compat+0x184/0x1c0
[ 4930.406785] Code: aa1303e0 97ffff18 f9408662 17ffffea (d4210000)
[ 4930.409485] ---[ end trace 441238610fd6c382 ]---
[ 4930.411782] Kernel panic - not syncing: Oops - BUG: Fatal exception in interrupt
[ 4930.426144] SMP: stopping secondary CPUs
[ 4945.228912] SMP: failed to stop secondary CPUs 1-2
[ 4945.234436] Kernel Offset: disabled
[ 4945.236120] CPU features: 0x0240002,24002004
[ 4945.237083] Memory Limit: none
[ 4945.242589] Rebooting in 1 seconds..
[ 4946.311770] SMP: stopping secondary CPUs
[ 4955.989754] SMP: failed to stop secondary CPUs 1-2
[ 4956.000053] Reboot failed -- System halted`

This never shows up before. Any stable branch i can use now?

jpfr, 2023-01-25:
I don’t think whatever we do can be responsible for a kernel panic. Sounds like a problem with the emulated peripherals…

eziosella, 2023-12-19:
Dear all, I attach on this previous issue because I am following my implementation on this same machine. Now my application is runnin as server as Euromap77 (not fully implementented on xml instances). Anyway as described on my CMAKE parameters the server is running with -DUA_MULTITHREADING=0.
If I connect from UAExpert everything seems working fine, reading and writing parameters without problem.
But using a thirdy part MES connecting, the server crash with the attached log gdb backtrace.
My last code calling is

> 20 0x0855df97 in beforeReadTime (server=0xb098e228, sessionId=0xb098e5b8, sessionContext=0x0, nodeid=0xa904a8b0, nodeContext=0x0, range=0x0, data=0xa904a960) at formctrlsoptions.cpp:341

Which is a simple UA_Server_writeValue where a node has been updated from a callback.onRead function.

My doubt is, could it depends from multithreading disable? Or from the backtrace do you read any other reasons?

Thanks,
Bye 
Ezio
[myLog.log](https://github.com/open62541/open62541/files/13714972/myLog.log)