open62541/open62541 #5573: CMake problem

Phil0707, 2023-01-16:
## Description
I can't build another project with open62541 because i get the following error using cmake ..
I need a description how i can include Events. Thanks in advance!

## Background Information / Reproduction Steps
This is the link to the other project https://github.com/opcua-skills/plug-and-produce.

Used CMake options:

```bash
build$ cmake ..
CMake Warning at /usr/local/lib/cmake/open62541/open62541Config.cmake:53 (message):
  Unsupported open62541 component: Events
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

CMake Error at CMakeLists.txt:22 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/open62541/open62541Config.cmake

  but it set open62541_FOUND to FALSE so package "open62541" is considered to
  be NOT FOUND.

-- Configuring incomplete, errors occurred!
..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): commit : [dac2c98d3b9722fb1971780d34996ea64bf65445](https://github.com/open62541/open62541/commit/dac2c98d3b9722fb1971780d34996ea64bf65445)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-01-16:
@Pro , you are active here and the author of the plug&produce paper.
If you can have a look.

JohannesKauffmann, 2023-05-25:
@Phil0707 For the `plug-and-produce` project, it looks like you'll need to build open62541 with `-DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON`, as it requires some extra components not enabled in open62541 by default.

The relation between the CMake flags and the CMake components comes [from here](https://github.com/open62541/open62541/blob/7d602cb1eb5bbf27ab7ad7ab85f5b617266110ca/CMakeLists.txt#L1683).

marwinglaser, 2025-03-25:
The installation no longer uses components, so this should be resolved.

closed 2025-03-25 by marwinglaser