open62541/open62541 #5578: generate_datatypes running error

eziosella, 2023-01-17:
## Description
Hi all,
following the instructions for XML Nodeset compiler:

>https://www.open62541.org/doc/0.3/nodeset_compiler.html

I am facing this error:

```bash
:~/open62541/tools$ python3 ./generate_datatypes.py --namespace=1 --type-csv=../deps/ua-nodeset/DI/Opc.Ua.Di.NodeIds.csv --type-bsd=../deps/ua-nodeset/DI/Opc.Ua.Di.Types.bsd --no-builtin ../build/src_generated/ua_types_di
Traceback (most recent call last):
  File "./generate_datatypes.py", line 89, in <module>
    [idx, ns] = m.split(':', 1)
ValueError: not enough values to unpack (expected 2, got 1)
```

Probably it sounds silly for many people more expert, but I do not know how to proceed with this.
BTW:
>${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/OpcUaDiModel.csv

no longer exists in the current project and I replace it with "Opc.Ua.Di.NodeIds.csv" instead, is it correct?

## Background Information / Reproduction Steps
#5570 here there are my CMAKE informations

Used CMake options:

```bash
cmake -DCMAKE_INSTALL_PREFIX=/opt/open62541 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=Full -DBUILD_SHARED_LIBS=On -DUA_MULTITHREADING=0 -DUA_FORCE_WERROR=Off ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): f4db84ae8b0a267a4a2be6ebe445d9832640e583
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 18.04.6 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-01-17:
You use the documentation for the 0.3 branch.
Try the master documentation if you use such a recent commit: https://www.open62541.org/doc/master/nodeset_compiler.html

Some of the CSV files have changed their location in the repository of the OPC Foundation.
Sometimes the git submodule is not updated even though open62541 has updated the reference.
Try to reset the git submodule to the version that is linked to from the commit you are using:
https://gist.github.com/nicktoumpelis/11214362

eziosella, 2023-01-17:
Sorry, my mistake.
Following the proper documentation version, so far so good.
The cleans and resets a git repo and its submodules was quite important.
Thanks!

JohannesKauffmann, 2023-05-25:
@jpfr This issue can be closed.

closed 2023-10-20 by eziosella