open62541/open62541 #5579: Solution for export OPC UA to XML like as in open62541-nodeset-loader.

xydan83, 2023-01-17:
In the open62541 was integrated open62541-nodeset-loader for importing XML structure to nodeset. And I have a question, will there be some solution to export OPC UA to XML?

vvpnet, 2023-01-18:
I also want to have this functionality!

xydan83, 2023-03-14:
Well, I'm preparing this solution, but it will be a C++ library that will use open62541. When it is done, I will write about it and put it on github.

vvpnet, 2023-03-20:
> Well, I'm preparing this solution, but it will be a C++ library that will use open62541. When it is done, I will write about it and put it on github.
Hi! I wish you good luck! I hope it won't take a lot of free time! Thanks!

vvpnet, 2023-04-07:
> Well, I'm preparing this solution, but it will be a C++ library that will use open62541. When it is done, I will write about it and put it on github.
Hi. How are you? Is everything working out? Maybe you need help in testing?

xydan83, 2023-04-07:
> > Well, I'm preparing this solution, but it will be a C++ library that will use open62541. When it is done, I will write about it and put it on github.
> 
> Hi. How are you? Is everything working out? Maybe you need help in testing?

Hi! The process is underway. Not as fast as I'd like, but it's going.

I have three main implementation points.
1) The core that will manage the process of encoding to XML (or something else) and take data from the Server (directly or through the Client);
2) Implementation of the "Client (Server) <--> Core" interface. These are a few methods that help get some data through the library (like attributes, references, namespaces, aliases, etc.);
At the moment I made the implementation through the client.
3) Implementation of the "XML Encoder <--> Core" interface. An XML encoder is one of the hardest things to get right.

By the way, the core connects to the other two components via an interface, which will make it possible to use your own implementations. It's usually accepted to encode the nodeset in CML, but maybe someone wants to use JSON?)

Now I have finished 1 and 2 points and am working on the 3rd one.
It's not fast, and I think I'll need some help with testing and maybe developing the project, because working with the OPS protocol doesn't seem easy)).

Well, when I finish the first version of the project, I will put it on github, and everyone can try it.

xydan83, 2023-04-07:
You can read about XML element types here: https://reference.opcfoundation.org/Core/Part6/v104/docs/F

(I'm using protocol 1.04 because that's the most common version these days, despite being certified under version 1.03).

vvpnet, 2023-04-10:
> Hi! The process is underway. Not as fast as I'd like, but it's going.
>
> I have three main implementation points.

I am very glad that you took up this difficult and difficult work!

1. I think it should be implemented as a plugin in the server. 

2. I don't think it's a good idea if the encoding process is launched synchronously in the server with a client call. Imagine that you have 5M nodes and the client wants to get them? There is only one event processing thread, it cannot encode in a small amount of time. It's better not to do it then. Or do it asynchronously.

3. Yes, you're right, this is the most difficult work! And here you also need compatibility with the nodeset-loader plugin.

JSON is good, but then you also need nodeset-loader-json.

I am ready to assist in testing.

I am waiting and wish you good luck!

xydan83, 2023-04-10:
> I am very glad that you took up this difficult and difficult work!
>
> 1. I think it should be implemented as a plugin in the server.
> 2. I don't think it's a good idea if the encoding process is [...]

Hi!
1) This can be done after development. Also, I think I can't add my library to the Open62541 project because C++. But in my fork I can add my addon as a plugin like nodeset-importer;

2) If you look at synchronous functions like **__UA_Client_Service** and then **sendRequest** you will see that all functions are based on 'asynchronous principles'. In this library, asynchronous functions differ from synchronous functions only in that in synchronous functions your calling function will wait, but the Client thread will not stop since already your call in the loop will spin the EventLoop. And the Client can do his job.
You can compare this function with **__UA_Client_AsyncServiceEx**.

And in addition from the documentation: _All OPC UA services are asynchronous in nature._
https://www.open62541.org/doc/1.3/client.html#asynchronous-services

For example, sendRequest Functions:
[image]

In this code, underarm, you will see wait operation for sync-functions **__UA_Client_Service**
[image]

I'm not sure yet about running such functions from another thread, since the Client's run cycle will be pushed by the request function, but in one thread everything should be fine and asynchronous.
By the way, not so long ago, client functions became thread-safe, just like server ones.

https://github.com/open62541/open62541/blob/7e456cb77b25d68649f0d68afc852eb09ebbdb26/CHANGELOG
```
2022-11-19 Julius Pfrommer <julius.pfrommer at iosb.fraunhofer.de>
 * Thread-safe client

   A large portion of the client API is now marked UA_THREADSAFE. For
   those methods, if multithreading is enabled, an internal mutex
   protects the client.
```

3) No, I will not write my own JSON encoder, but if someone wants to, they can implement the encoder interface in their own way. And you are right, you will need some kind of **nodeset_json_importer** for this). But in most cases, exporting or importing a node set is done via XML.

Thanks for your help in the future).

vvpnet, 2023-06-01:
> 
> Thanks for your help in the future).

Hi! How are you doing?

xydan83, 2023-06-02:
> > Thanks for your help in the future).
> 
> Hi! How are you doing?

Hi.

Step by step.
I've done the main part of my work and am writing some unit tests.

[image]
Screenshot from the IDE with passing tests.

And this is the first generation of xml from the OPC UA Server with little structure but all the main node classes are present.

```typescript
<?xml version="1.0" encoding="UTF-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
    <NamespaceUris>
        <Uri>urn:open62541.server.application</Uri>
        <Uri>http://devnetiot.com/opcua/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="EnumValueType">i=7594</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="Int64">i=8</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="SamplingIntervalDiagnosticsDataType">i=856</Alias>
        <Alias Alias="String">i=12</Alias>
    </Aliases>
    <!--Value elements are currently not supported in Variable and VariableType nodes.-->
    <!--Definition elements are currently not supported in UADataType.-->
    <UAObject NodeId="ns=2;i=1" BrowseName="2:vPLC1" ParentNodeId="i=85">
        <DisplayName>vPLC1</DisplayName>
        <Description>Description vPLC1</Description>
        <References>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=5</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=4</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=3</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=22</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=2</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=30</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=29</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=28</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=24</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=23</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=21</Reference>
        </References>
    </UAObject>
    <UAObject NodeId="ns=2;i=5" BrowseName="2:myNewStaticObject1" ParentNodeId="ns=2;i=1">
        <DisplayName>myNewStaticObject1</DisplayName>
        <Description>Description myNewStaticObject1</Description>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=2;i=1</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=9</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=8</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=7</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=6</Reference>
            <Reference ReferenceType="HasComponent">ns=2;i=17</Reference>
        </References>
    </UAObject>
    <UAVariable NodeId="ns=2;i=4" BrowseName="2:pumpsetting" ParentNodeId="ns=2;i=1" DataType="String" ValueRank="-2">
        <DisplayName>pumpsetting</DisplayName>
        <Description>Description pumpsetting</Description>
[... 197 lines cut ...]
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        </References>
    </UAVariable>
    <UADataType NodeId="ns=2;i=32" BrowseName="KeyValuePair_test">
        <DisplayName>KeyValuePair_test</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=31</Reference>
        </References>
    </UADataType>
    <UAVariable NodeId="ns=2;i=16" BrowseName="2:MyProperty2" ParentNodeId="ns=2;i=15" DataType="String" ValueRank="-2">
        <DisplayName>MyProperty2</DisplayName>
        <Description>Description MyProperty2</Description>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=2;i=15</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
    </UAVariable>
</UANodeSet>
``` 

This xml passed SCHEMA (UANodeSet.xsd) validation and python **nodeset_compiler** validation and the Open62541 server can load it.

[image]
In this screenshot, you can see the test structure that this xml data was made from.

vvpnet, 2023-06-05:
This is a very good result! Is there any expectation of the timing of the first option ?

xydan83, 2023-06-08:
> This is a very good result! Is there any expectation of the timing of the first option ?

Will be finished when the time comes)

vvpnet, 2023-09-21:
> > This is a very good result! Is there any expectation of the timing of the first option ?
> 
> Will be finished when the time comes)
Hi! How are things going with you ? There has been no news for a long time :)

vvpnet, 2023-10-18:
> > This is a very good result! Is there any expectation of the timing of the first option ?
> 
> Will be finished when the time comes)

Hi! Where have you disappeared to? Tell me, did you manage to finish export?

vvpnet, 2023-10-19:
> Hello! Yes, the main part of nodesetexporter is complete. I have some problems and the main one is that I can't export custom node types and values, it takes a lot more time, but the generic node type can be exported.
> But I don't know when I will be able to release the project to git, because it is part of a working project, but I have agreed with my employer that I can make the code publicly available.

Many human thanks to you for trying to do for all interested people in this. And make this functionality open. I think after the publication of the functionality, some problems will be solved by the community, including I want to participate in this. I'm looking forward to it :)

xydan83, 2023-10-19:
> Many human thanks to you for trying to do for all interested people in this. And make this functionality open. I think after the publication of the functionality, some problems will be solved by the [...]

For standalone use I have created CLI part, for example you can set the opc ua server endpoint and start node and get a set of .xml nodes.
I believe that my employer will soon give me the opportunity to promote this project on git. Before this, I solve many issues alone.

vvpnet, 2023-10-20:
> > > Hello! Yes, the main part of nodesetexporter is complete. I have some problems and the main one is that I can't export custom node types and values, it takes a lot more time, but the generic node type can be exported.
> > > But I don't know when I will be able to release the project to git, because it is part of a working project, but I have agreed with my employer that I can make the code publicly available.
> > 
> > 
> > Many human thanks to you for trying to do for all interested people in this. And make this functionality open. I think after the publication of the functionality, some problems will be solved by the community, including I want to participate in this. I'm looking forward to it :)
> 
> For standalone use I have created CLI part, for example you can set the opc ua server endpoint and start node in linux console and get a set of .xml nodes. I believe that my employer will soon give me the opportunity to promote this project on git. Before this, I solve many issues alone.

CLI part is a very good style and correct. Whenever you do something alone, you're never sure if it's right. I have sent you an email to the email address specified in your profile.

xydan83, 2023-10-23:
> CLI part is a very good style and correct. Whenever you do something alone, you're never sure if it's right. I have sent you an email to the email address specified in your profile.

Hello!
As I wrote, I made this nodesetexporter (I called it that :)) for the company where I work and each of our code passes CR, but I made this program according to the standard, based on the UANodeSet.xsd scheme. In other words, there is validation according to this scheme. Well, as I wrote, unfortunately, placing code in Git depends not only on me. But I agreed and hope that the employer will not go back on his words.

vvpnet, 2023-10-26:
OK, everyone interested is waiting :)

xydan83, 2023-11-03:
@jpfr Hi!
I'm preparing an open source (free) solution to export a set of nodes to xml. This is similar to the '**open62541-nodeset-loader**' project, but I'm using **C++**. Can I name this project (repository) as "**open62541-nodeset-exporter**"? 
And, in the future, I want to connect this project to your open62541 repository, and with integration as a plugin. if it possible?

jpfr, 2023-11-03:
Hey there,

you can use that name for your repository.
So far we don’t have C++ code in our project.
But it can be a possibility to move it over.

Please release this first as-is.
And then we can see what the advantages and requirements are going forward.

See you, Julius

xydan83, 2023-11-03:
> Hey there, you can use that name for your repository. So far we don’t have C++ code in our project. But it can be a possibility to move it over. Please release this first as-is. And then we can see [...]

Thanks for the answer. At the beginning I wrote my message in this issue, but after that I thought it would be better to write it in a new issue, but now this is not necessary. You have already answered here. :)
I'm understood, thank you!
Once I've created and posted my code, I'll create a new issue with a link to my repo.

xydan83, 2024-01-22:
@vvpnet @jpfr 
Hi!
I finally finished the first version of the nodeset exporter to xml and uploaded the code to my repositories. Please check out the project.
https://github.com/xydan83/open62541-nodeset-exporter

@jpfr Since the project is written in C++, I will make a wrapper for C compatibility, but still, during assembly, the code will require a C++ compiler.
What are the requirements to be able to include a nodeset exporter in your library Open62541 as a plugin, similar to [open62541-nodeset-loader](https://github.com/open62541/open62541-nodeset-loader)?

vvpnet, 2024-01-23:
@xydan83 
Hello! Thank you so much! For your work done! I will start testing your solution in the near future. I hope she can handle my demands. I have  UA servers with the number of nodes > 2000000 :)

xydan83, 2024-01-23:
> @xydan83 Hello! Thank you so much! For your work done! I will start testing your solution in the near future. I hope she can handle my demands. I have UA servers with the number of nodes > 2000000 :)

WoW! Pretty cool)) I will be waiting report)) Very interested))