open62541/open62541 #5601: PubSub

HaraldEschbach, 2023-01-27:
## Description

## Background Information / Reproduction Steps
We use OPC UA server with PubSub on VxWorks and need to support security for it. 
VxWorks is using openSSL to encryption other protocols
Also open62541 has already some support for openSSL but not for PubSub.
It does not realy make sense to use two different libs on an device.

## Checklist

Please provide the following information:

 - [ ] open62541 Version 1.3:
 - [ ] Operating system: VxWorks

jpfr, 2023-01-27:
Yes. It would be great to have PubSub security also for OpenSSL.
This hinges on these two files:

https://github.com/open62541/open62541/blob/master/plugins/crypto/mbedtls/securitypolicy_pubsub_aes128ctr.c
https://github.com/open62541/open62541/blob/master/plugins/crypto/mbedtls/securitypolicy_pubsub_aes256ctr.c

Porting them to OpenSSL should be straightforward.
Especially since most SecurityPolicy plugins are structurally the same.
Only a few key lengths and/or crypto methods from the underlying library change.

I'd estimate this requires 1-3 days of work to port.

Muddasir-Shakil, 2023-01-30:
@jpfr 
can you add label here for help wanted and good first issues.?