open62541/open62541 #5611: BUG: Client: Browsing or subscribing failed with latest ZipTree updates

GoetzGoerisch, 2023-02-01:
## Description

The latest ZipTree fix [a85bcee](https://github.com/open62541/open62541/pull/5580/commits/a85bcee25827707f4f2f23123b6cc14dd596bd3f) breaks our existing client implementation.

Error:
```
[DashboardOpcUaClient] INFO bool Umati::MachineObserver::MachineObserver::canBrowseMachineList(std::__cxx11::list<ModelOpcUa::BrowseResult_t>&):129 Searching for machines
Assertion failed: !UA_TimerTree_ZIP_MIN(&t->tree) || UA_TimerTree_ZIP_MIN(&t->tree)->nextTime > nowMonotonic (/src/DashboardOpcUaClient/deps/open62541/arch/common/ua_timer.c: UA_Timer_process: 261)
```
while the client starts browsing for machines.
Sometimes it only breaks later while starting to subscribe to nodes.

## Background Information / Reproduction Steps

Build container image: 
https://github.com/umati/Dashboard-OPCUA-Client/commit/dc5f2e3f1c7c395c8c3b8783467f6285d36b4eec

Use `configuration.json.example` and set the OPC UA endpoint to

```
  "OpcUa": {
    "Endpoint": "opc.tcp://opcua.umati.app:4840",
    "Username": "",
    "Password": "",
    "Security": 1
  },
```

and a available MQTT broker.
then run with
`docker run -it --rm -v /path/to/configuration.json:/app/configuration.json dashboardopcuaclient`

Used CMake options:

```bash
cmake -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
               -DUA_NAMESPACE_ZERO:STRING=FULL -DUA_ENABLE_ENCRYPTION:STRING=LIBRESSL
               -DLIBRESSL_ROOT_DIR:PATH=${CMAKE_INSTALL_PREFIX} -DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_INSTALL_PREFIX}
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): [f70693b](https://github.com/open62541/open62541/tree/f70693b08d097d2a49354903034e2a2029adc821)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Alpine-Linux 3.17
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-01:
Yikes. Will have a look.

Muddasir-Shakil, 2023-02-01:
A bit of background from my side on the issue.
I also had the same assertion only when I ran the (sks) unit tests with valgrind (mem checks enabled) enabled. The problem with the error was it was always sporadic. While running without valgrind (mem checks disabled) everytime the tests ran without any issues.

GoetzGoerisch, 2023-02-13:
@jpfr happens so far only on `alpine-3.17`

GoetzGoerisch, 2023-02-13:
Had to go back to fface1580a2ad8c9fbd1f87971c1900f875c29f5 to don't have it on alpine.

jpfr, 2023-02-13:
We are still on it.
See https://github.com/jpfr/open62541/blob/ziptree_proof/deps/ziptree_proof.c .
This is a "fully typed" version of the binary tree algorithm. For better analysis.

That thing is very much non-obvious.
In order to fully lock it down, we have to pull in some formal model checking/static validation/proof verifier.

CBMC and IKOS didn't work out.
Either couldn't validate the Assertions or took forever to run.
- https://www.cprover.org/cbmc/
- https://github.com/NASA-SW-VnV/ikos

Next up is Frama-C:
https://frama-c.com/

jonathan-ibex-controls, 2023-02-27:
I encountered this issue running the [Server Events Tutorial](https://www.open62541.org/doc/1.3/tutorial_server_events.html) example code, using Windows 10 and Visual Studio 2019.

Here is the error message:
`Assertion failed: !UA_TimerTree_ZIP_MIN(&t->tree) || UA_TimerTree_ZIP_MIN(&t->tree)->nextTime > nowMonotonic, file C:\Ibex\modbus-messenger\open62541\arch\common\ua_timer.c, line 262`

Here is my output:
[output.txt](https://github.com/open62541/open62541/files/10835131/output.txt)
Here is a .zip of my project:
[ibex-opc.zip](https://github.com/open62541/open62541/files/10835139/ibex-opc.zip)

jpfr, 2023-03-01:
A fix is coming up in https://github.com/open62541/open62541/pull/5660.
Unfortunately we have unrelated issues with the CI.
Jobs don't start atm...

GoetzGoerisch, 2023-03-01:
Saw this already! Great news.

jpfr, 2023-03-03:
Merged. Please check on your end and close/

GoetzGoerisch, 2023-03-06:
Sadly, still broken on my side at c0a36d0950f059d4e2052b19b83a0358ba0c17f8

GoetzGoerisch, 2023-03-30:
Thank you @jpfr ! Fixed in the latest main. 
Possible to point out the specific commit?

closed 2023-05-02 by GoetzGoerisch