open62541/open62541 #5614: feat(plugin):  plugin design to move plugin objects

Muddasir-Shakil, 2023-02-01:
hello everyone,
This issue is created to start the discussion on topic of making the deep copy/moving plugin objects. Recently UA_ClientConfig_copy is added to upstream, which makes  a partial deep copy of it members. The problem is caused by the plugins pointers such as security pointers, eventloop, loggers etc. The implementation of these plugins contains internal objects whose structs are not exposed outside of the plugin implementation, which makes it difficult (impossible) to make the deep copy of such objects also including serverconfig. Therefore, some rethinking is required to fix the issues in the plugins, so that they take care of making deep copies of their internally contained objects. There are various approaches can be taken to solve the above mentioned problem. I would like to suggest the following solutions and invite you to propose something else or comment on which approach is better. We can then create a PR to fix this issues in our plugins.

**suggestion**

1. Copy/move the pointers and keep record of reference counting technique inside the plugin implementation [(reference counting technique)](https://mortoray.com/what-is-reference-counting/)
2. Marking the object of the plugin by a flag as external, when moved/copied as done in the Eventloop implementation (ExternalEventloop)
3. Adding a new "copy" function to the Plugin API. The copy function takes care of the making the deep copy of the internal objects to the destination.

I suggest to use solution 1. The benefit of it as single object can be linked to multiple objects which means less complex logic to allocate/deallocate the memory. if the more number of copies are required which can be simply created without allocating any new memory, rather reuse the pointer to the original object.

Looking forward to your suggestions.
BR
Muddasir Shakil

jpfr, 2023-02-03:
I prefer option 2.
It allows more „manual control“ that is deterministic on when a plug-in is cleaned up.

In nearly all cases the plugins will „live“ for the duration of the entire application.
Either because they „live“ in a master-server/client.
Or because they are initialized in a global variable from which runtime-server/client-instances can access them during initialization.
And when the plug-ins are held in a global variable then it is easy to clean them up right before the application shuts down.

jpfr, 2023-02-03:
Option 2 also allows the re-linking you describe.
But there is one master-pointer that has a lifecycle longer than the other users.

Any other arguments we forgot?

Muddasir-Shakil, 2023-02-06:
the problem with option 2 is that the when we make a copy of the object the copied object can survive until the original survives. It will cause issues if the user decides to free the original object at runtime. If we also mark original object as external then it is difficult to differentiate between original and copied object when object is being cleaned up. 
Also with "manual control" has to be done additionally by the user, which may invite user to make mistakes.
I am in favor of option 2 because we can achieve similar behavior of  option 2, without exposing anything to the user and let the stack manage the copies internally. So the user can call for example UA_ClientConfig_clean without doing any other logic before. 
let me know what do you think? if you still feel option 2 is better approach then I can implement it using option 2.
BR
Muddasir Shakil

jpfr, 2023-02-15:
Ok, these are good arguments.
However, overall, I still prefer option 2.
There are plugin implementations that tightly bound to the surrounding application.
For example AccessControl plugins that connects to a backend service.

There should be no surprises when a plugin is deleted.
With option 2 we have the possibility that all plugin-users have the "external bit" set.
So the application has the full control on the lifecycle. Even when there are zero remaining users.