open62541/open62541 #5617: undefined reference to UA_[Variable] and UA_TYPES in main.c and myServer.c during compilation of nodeset

tedwad, 2023-02-03:
## Description
I am trying to run an open62541 server on a raspberry pi 4 using a custom nodeset. I created the information model following the instructions of the tutorial at [opcua.rocks](https://opcua.rocks/custom-information-models/). I successfully compiled my nodeset Opc.Ua.Ff.NodeSet2.xml (which has dependencies on the DI and Robotics companion specs) using the UA-ModelCompiler. I then ran the following script to generate the source files myServer.c and myServer.h for my model:

`python ~/open62541/tools/nodeset_compiler/nodeset_compiler.py --types-array=UA_TYPES -e ~/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml -e nodesets/DI/Opc.Ua.Di.NodeSet2.xml -e nodesets/Robotics/Opc.Ua.Robotics.NodeSet2.xml -x model/Published//Opc.Ua.Ff.NodeSet2.xml myServer`

I then tried to compile the server:
`gcc -std=c99 -I$HOME/install/include -L$HOME/install/lib main.c myServer.c -lopen62541 -lmbedtls -lmbedx509 -lmbedcrypto -o myServer`

which resulted in this error message:
[error_message.txt](https://github.com/open62541/open62541/files/10580620/error_message.txt)

I also tried to compile the server with the CMake Macro suggested method from the tutorial, but this resulted in the same error message. 

Here is a copy of my repository [conneXt-main.zip](https://github.com/open62541/open62541/files/10580637/conneXt-main.zip) if anyone wants to take a look at my code. Any help would be much appreciated.

## Background Information / Reproduction Steps

Used CMake options:

```bash
BUILD_SHARED_LIBS: OFF
CLANG_FORMAT_EXE: CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE: RelWithDebInfo
CMAKE_INSTALL_PREFIX: /home/ubuntu/conneXt/install
UA_ARCHITECTURE: posix
UA_BUILD_EXAMPLES: OFF
UA_BUILD_TOOLS: OFF
UA_BUILD_UNIT_TESTS: OFF
UA_ENABLE_AMALGAMATION: OFF
UA_ENABLE_DA: ON
UA_ENALBE_DIAGNOSTICS: OFF
UA_ENABLE_DISCOVERY: OFF
UA_ENABLE_ENCRYPTION: OFF
UA_ENABLE_ENCRYPTION_TPM2: OFF
UA_ENABLE_HISTORIZING: OFF
UA_ENABLE_JSON_ENCODING: ON
UA_ENABLE_METHODCALLS: ON
UA_ENABLE_NODESETLOADER: OFF
UA_ENABLE_PUBSUB: ON
UA_ENABLE_PUBSUB_DELTAFRAMES: OFF
UA_ENABLE_PUBSUB_ETH_UADP: OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL: OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS: OFF
UA_ENABLE_SUBSCRIPTIONS: ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS: ON
UA_ENABLE_XML_ENCODING: OFF
UA_LOGLEVEL: 100
UA_MULTITHREADING: 0
UA_NAMESPACE_ZERO: FULL
UA_NODESET_DIR: /home/ubuntu/open62541/deps/ua-nodeset
```

## Checklist

Please provide the following information:

 - [x] open62541 Version 1.3.4:
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04.5
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-06:
All the mentioned symbold are exported by the library. So **-lopen62541** should bring them in.
Did you do a `make install` of the library in the recent version?

tedwad, 2023-02-06:
Yes, I configured the CMake options (shown above) and then built the library using the following commands:
```
cd ~/open62541
mkdir build && cd build
ccmake ..
make -j
make install
```

jpfr, 2023-02-08:
Read the namespaces variable under the server object to see which namespace was assigned to which id.

tedwad, 2023-02-08:
Where can I find the server object?

closed 2023-04-24 by tedwad