open62541/open62541 #5622: Can't compile on Windows 11: undefined reference to 'if_nametoindex'

KnowingBat, 2023-02-09:
## Description
I recently installed open62541 on Windows 11 and build it successfully. I used ```amalgamation``` in order to use the open62541.c/.h files. I setup the example project called ```tutorial_server_firststeps.c``` and, as i'm using a MinGW-64 compiler, i tried to compile as mentioned in the documentation, using the command ```gcc -std=c99 open62541.c tutorial_server_firststeps.c -lsw2_32 -o test.exe ```

## Background Information / Reproduction Steps
Running the command above, i bump into this error:
[image]

This is my current version of MinGW-64:
[image]

I also tried to change MinGW-64 from ```posix``` to ```win32```, but the issue still remains.

In addiction, this is my "tutorial_server_firststeps.c" file:
[image]

Is there something that i'm missing here? 

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-15:
Hmm. After creating the open62541.c.h files, I can only find benign uses:
```
jpfr@virtualbox:~/software/open62541/build$ grep -r -H nametoindex
open62541.h:# define UA_if_nametoindex if_nametoindex
open62541.h:# ifndef UA_if_nametoindex
open62541.h:unsigned int UA_if_nametoindex(const char *ifname);//equivalent to posix if_nametoindex implementation
```

Please do the same to see where this comes from.

KnowingBat, 2023-02-16:
Hello @jpfr, i did what you suggested, this is the output:

```  
open62541.c: ipMulticastRequest.ipv6.ipv6mr_interface = UA_if_nametoindex(interfaceAsChar);
open62541.c: ipMulticastRequest.ipv6.ipv6mr_interface = UA_if_nametoindex(interfaceAsChar);
open62541.h:# define UA_if_nametoindex if_nametoindex
open62541.h:# define UA_if_nametoindex if_nametoindex
open62541.h:# ifndef UA_if_nametoindex
open62541.h:unsigned int UA_if_nametoindex(const char *ifname);//equivalent to posix if_nametoindex implementation 
```
Actually i did "solve" the problem by commenting out ```#define UA_IPV6 1```. That's fine because i don't really need IPV6 right now. However i don't really understand why if_nametoindex is causing this issue...  Do you have any further informations about this?

closed 2023-03-14 by KnowingBat

8188, 2023-07-21:
Add **-lIphlpapi** will solve this problem. The reason is [here.](https://learn.microsoft.com/en-us/windows/win32/api/netioapi/nf-netioapi-if_nametoindex)
At last, with MinGW-64 on Windows10 my command line: 
`gcc tutorial_server_firststeps.c -o tutorial_server_firststeps -I..\build\src_generated -I..\include -I..\arch -I..\deps -L..\build\bin -lopen62541 -lws2_32 -lIphlpapi`