open62541/open62541 #5627: Invoking methods with no arguments from open62541 based client

jknudstr, 2023-02-09:
Referring to 

https://github.com/FreeOpcUa/opcua-asyncio/discussions/1197

it seems that there is a small discrepancy from the OPC UA standard in open62541, on the client side, when it comes to invoking methods, having no arguments, in the associated server.

Would be great to have this fixed!

GoetzGoerisch, 2023-02-09:
Please use the issue template to be able to reproduce this for the developers.

jknudstr, 2023-02-10:
OK, will do.

danmoser, 2023-02-22:
@GoetzGoerisch, I'm facing the exactly same issue and it is pretty simple to reproduce: you just need a method without arguments and called it from an open62541 and compare it with call from [UaExpert](https://www.unified-automation.com/products/development-tools/uaexpert.html) of from [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio).

This is the log from an opcua-syncio server being called by UaExpert:
```
Calling: CallMethodRequest(ObjectId=NodeId(Identifier='NodeIsString', NamespaceIndex=4, 
NodeIdType=<NodeIdType.String: 3>), MethodId=NodeId(Identifier='NodeIsString#noArgMethod', 
NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>), InputArguments=[])
```

Same server being called by open62541:
```
Calling: CallMethodRequest(ObjectId=NodeId(Identifier='NodeIsString', NamespaceIndex=4, 
NodeIdType=<NodeIdType.String: 3>), MethodId=NodeId(Identifier='NodeIsString#noArgMethod', 
NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>), InputArguments=None)
[ERROR] Error executing method call CallMethodRequest(ObjectId=NodeId(
Identifier='NodeIsString', NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>), 
MethodId=NodeId(Identifier='NodeIsString#noArgMethod', NamespaceIndex=4, 
NodeIdType=<NodeIdType.String: 3>), InputArguments=None), an exception was raised: 
Traceback (most recent call last):
  File "/base/lib/python3.8/site-packages/asyncua/server/address_space.py", line 580, in _call
    result = await self._run_method(node.call, method.ObjectId, *method.InputArguments)
TypeError: _run_method() argument after * must be an iterable, not NoneType
```

jknudstr, 2023-03-30:
I have to apologize for not having provided the proper information according to the template for error reporting.
What happens is that very shortly after i reported the issue here, a patch (work around) was implemented in FreeOpcUa/opcua-asyncio, which resolved this issue for me. This patch was quickly installed on our development machines, so that I cannot reproduce the problem anymore.
I _do_ have a small client/server example, which in principle would reproduce the issue, with the previous version of FreeOpcUa/opcua-asyncio. I can give this to you if you like.
Apart from that, I think I have to leave it up to you to decide whether to fix this, apparent discrepancy in open62541.
As mentioned by @danmoser it is quite straightforward to reproduce.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The cause was in the peer stack’s client and was fixed there (see [FreeOpcUa/opcua-asyncio#1217](https://github.com/FreeOpcUa/opcua-asyncio/issues/1217)), so nothing is left to change in open62541 and we are closing this.

closed 2026-10-20 by o6-nbeier