open62541/open62541 #5631: The client_encryption.c example program fails with mbedTLS 3.3.0

robert-johnston, 2023-02-10:
## Description
Running the client_encryption.c example program fails.
I'm connecting to a commercial UA server and when I use the same certificate from a commercial UA client connecting to the same server the connect succeeds so I believe the certificates I'm using are good.

## Background Information / Reproduction Steps
Create a solution in Visual Studio 2022.
Create a C console project using client_encryption.c, open62541.c and open62541.h.
Modify the client_encryption.c file to include the loadFile() function and use the amalgamated header.
Build the mbedtls 3.3.0 project from source.
Include the mbedTLS project in the VS solution and reference it in the client_encryption project.
In the client_encryption project properties point the include directorys and included libraries to the appropriate locations in mbedTLS project build folder.
Generate ssl certificates.  I used openssl in Linux.
Set the URI in the SAN of the certificate to URI.1 = urn:unconfigured:application  and include the IP address of the client.

Run the client_encryption executable.
command line: 
client_encryption.exe opc.tcp://localhost:51310/Server  client.crt  client.key 

The certificate and key are in *.pem format with no passphrases.
The server supports Basic128rsa15, Basic256 and Basic256sha256.

I'm connecting to a commercial server and so I run client_encryption.exe once which transfers the certificate to the server.
Then on the server I put the certificate into the trusted certificate store.
Then run the client_encryption.exe again and it fails to produce the expected output giving instead the error message:
 "Receiving the response failed with StatusCode BadSecurityChecksFailed"

The problem seems to be that the client doesn't decode the response from the server.

Used CMake options:

```In Windows:
cmake .. -G "Visual Studio 17 2022" -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_AMALGAMATION=ON  -DUA_NAMESPACE_ZERO=FULL  -DUA_ENABLE_ENCRYPTION=MBEDTLS  -DMBEDTLS_INCLUDE_DIRS="C:\mbedtls-3.3.0\include\mbedtls" -DMBEDTLS_LIBRARY="C:\mbedtls-3.3.0\library\Release" -DMBEDX509_LIBRARY="C:\mbedtls-3.3.0\library\Release" -DMBEDCRYPTO_LIBRARY="C:\mbedtls-3.3.0\library\Release" -DBUILD_SHARED_LIBS=ON -D CMAKE_BUILD_TYPE=Release -DUA_LOGLEVEL=300
```

For the mbedTLS build:
cmake . -DUSE_SHARED_MBEDTLS_LIBRARY=ON -D CMAKE_BUILD_TYPE=Release

## Checklist
open62541 version is v1.3.4
no other SDKs
OS Windows 11
output attahced
wireshark trace attached
source file attached
script used to generate certificates attached

[output.txt](https://github.com/open62541/open62541/files/10709791/output.txt)
[client_encryption_trace.zip](https://github.com/open62541/open62541/files/10709803/client_encryption_trace.zip)
[encryptionTest.zip](https://github.com/open62541/open62541/files/10709809/encryptionTest.zip)
[gen2.zip](https://github.com/open62541/open62541/files/10709826/gen2.zip)

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ffreality, 2023-02-15:
1- MBedTLS doesn't support subject alternative names. Prosys like servers requires it.
Your client description app uri should be same with your certificate's URI. So, I suggest you to use OpenSSL.

2- you should define security mode

3- UA_CertificateVerification_AcceptAll after UA_ClientConfig_setDefaultEncryption

This is my connection code from Unreal.
FString is equal with const char*

`
void UOPC_ClientBPLibrary::OPC_UA_Client_Connect(FClientConnection DelegateClientConnection, const FString InAppName, const FString InAppUri, UCertificate* InCertificate, const FString InTrustListPath, const FString InRevokeListPath, const FString InAddress, const FString InUsername, const FString InPassword, ESecurityModes InSecurityModes, ESecurityPolicyURIs InSecurityPolicyUri, int32 InTrustListSize, int32 InRevokeListSize)
{
	AsyncTask(ENamedThreads::AnyNormalThreadNormalTask, [DelegateClientConnection, InAppName, InAppUri, InCertificate, InTrustListPath, InRevokeListPath, InAddress, InUsername, InPassword, InSecurityModes, InSecurityPolicyUri, InTrustListSize, InRevokeListSize]()
		{
			// Construct OPC UA Client object referance.
			UOPC_UA_Client* OPC_Client = NewObject<UOPC_UA_Client>();
			OPC_Client->Client = UA_Client_new();

			// Create client configurations.
			UA_ClientConfig* ClientConfig = UA_Client_getConfig(OPC_Client->Client);
			ClientConfig->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;
			ClientConfig->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", TCHAR_TO_ANSI(*InAppName));
			ClientConfig->clientDescription.applicationUri = UA_STRING_ALLOC(TCHAR_TO_ANSI(*InAppUri)); // Converter: UE FString to UA_String.

			//https://profiles.opcfoundation.org/profile/925
			switch (InSecurityPolicyUri)
			{
				case ESecurityPolicyURIs::None:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#None");
				break;
				
				case ESecurityPolicyURIs::Aes128_Sha256_RsaOaep:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");
				break;
				
				case ESecurityPolicyURIs::Basic256_Sha256:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
				break;
				
				case ESecurityPolicyURIs::Aes256_Sha256_RsaPss:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss");
				break;
	
				case ESecurityPolicyURIs::ECC_curve25519:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_curve25519");
				break;
	
				case ESecurityPolicyURIs::ECC_nistP256:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_nistP256");
				break;
	
				case ESecurityPolicyURIs::ECC_nistP384:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_nistP384");
				break;
				
				case ESecurityPolicyURIs::ECC_brainpoolP256r1:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_brainpoolP256r1");
				break;
	
				case ESecurityPolicyURIs::ECC_brainpoolP384r1:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_brainpoolP384r1");
				break;
	
				case ESecurityPolicyURIs::ECC_curve448:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#ECC_curve448");
				break;
				
				default:
				ClientConfig->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
[... 124 lines cut ...]
				
				AsyncTask(ENamedThreads::GameThread, [DelegateClientConnection, OPC_Client, Connection_Status]()
					{
						DelegateClientConnection.ExecuteIfBound(true, UA_StatusCode_name(Connection_Status), OPC_Client);
					}
				);
			}

			else
			{
				UA_Client_delete(OPC_Client->Client);
				
				UA_ByteString_clear(&Certificate);
				UA_ByteString_clear(&PrivateKey);

				AsyncTask(ENamedThreads::GameThread, [DelegateClientConnection, OPC_Client, Connection_Status]()
					{
						DelegateClientConnection.ExecuteIfBound(false, UA_StatusCode_name(Connection_Status), nullptr);
					}
				);
			}
		}
	);
}
`

robert-johnston, 2023-02-21:
*.Pem certificates will work with the client program it is not necessary to use *.der.

I did use open ssl and provided the script to create the certificates.

The certificate URI is "urn:unconfigured:application" which is the client default URI.  I used this to overcome the open62541 client certificate validation check that requires URI to match the application URI.  And the certificate does pass that check.

I want to run the example program as it is.  Has anyone else encountered problems with using recent Mbedtls and the example program?

The code worked previously with mbedtls 2.2.7.  But something has changed in open62541 so that it no longer functions.

robert-johnston, 2023-03-22:
The client_encryption example program only fails for connecting via Basic128rsa15 encryption connections.  The other encryptions algorithms decrypt by a different path in the client.  This seems to be a problem in the open62541 library not being compatible with mbedtls 3.3.0 as described below.

Here's what I've found.

When processing a Basic128rsa15 connection the open62541 client eventually calls function:
asym_decrypt_sp_basic128rsa15
That function calls function:
mbedtls_pk_decrypt  (which is an mbedtls function)  and it passes in NULL for the random number generator function pointer argument.
This is the cause of the malfunction when the mbedtls version is 3.3.0 and probably other mbedtls with version >= 3.0.

When mbedtls 3.3.0 processes the call to mbedtls_pk_decrypt it eventually calls function:
mbedtls_rsa_private  (found in file rsa.c)
In that function there is this check which fails because the function pointer f_rng passed in is NULL, coming from Open62541.
    if( f_rng == NULL )
        return( MBEDTLS_ERR_RSA_BAD_INPUT_DATA );

jpfr, 2023-03-22:
Thanks for the deep analysis.
Would you open an issue on GitHub for that?

Regards

jpfr, 2023-04-05:
Right. I had received this via mail. So it's already a Github issue.
Fixed via #5724.

JohannesKauffmann, 2023-05-25:
@jpfr This issue can be closed.

closed 2025-03-18 by marwinglaser