open62541/open62541 #5637: The server configuration struct UA_ServerConfig doesn't own a "void* context" pointer

hobbitdev, 2023-02-13:
## Description
The server configuration struct UA_ServerConfig doesn't own a "void* context" pointer useful to attach an "external" object.
Such kind of opportunity is present in UA_ClientConfig with clientContext member.
I would have espected something similar for UA_ServerConfig, but apparently it is missing.

On a server side, what's the way to attach an external  C++ object pointer in order to retrive it in UA_ValueCallback callback functions ?
Could someone point me to the right way to achive it?

Thanks in advance
Rossano

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -S'source path' \
	-B'build path' \
    -DOPENSSL_ROOT_DIR=$OPENSSLPATH \
	-DCMAKE_BUILD_TYPE=Release \
	-DBUILD_SHARED_LIBS=OFF \
	-DUA_NAMESPACE_ZERO=FULL \
	-DUA_ENABLE_SUBSCRIPTIONS=ON \
    -DUA_ENABLE_DISCOVERY=ON \
	-DUA_ENABLE_ENCRYPTION=OPENSSL \
	-DUA_BUILD_SELFSIGNED_CERTIFICATE=ON \
	-DUA_ENABLE_METHODCALLS=ON \
	-DUA_ENABLE_NODEMANAGEMENT=ON \
	-DUA_LOGLEVEL=300
```

## Checklist

Please provide the following information:

 - [x] open62541 Version 1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-15:
Implemented in https://github.com/open62541/open62541/pull/5640

jpfr, 2023-02-15:
Btw, this is already done on master.
Backporting to 1.3 now.

closed 2023-02-15 by jpfr