open62541/open62541 #5638: PubSub subscriber crashes when using value callbacks

JuhaSointusalo, 2023-02-14:
## Description

If a node has value callback set then subscriber crashes when attempting to call the `onWrite` callback.

```
[2023-02-14 17:25:41.591 (UTC+0200)] trace/eventloop    Iterate the EventLoop
[2023-02-14 17:25:41.592 (UTC+0200)] debug/pubsub       ReaderGroup ns=1;i=50448        | PubSub subscribe callback
[2023-02-14 17:25:41.596 (UTC+0200)] debug/pubsub       ReaderGroup ns=1;i=50448        | Reader ns=1;i=50450   | Process Msg with DataSetReader!
[2023-02-14 17:25:41.596 (UTC+0200)] debug/pubsub       ReaderGroup ns=1;i=50448        | Reader ns=1;i=50450   | DataSetReader 'DataSet Reader 1': received a network message
[2023-02-14 17:25:41.596 (UTC+0200)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=1;i=50000
tutorial_pubsub_subscribe: /media/projects/open62541/arch/posix/ua_architecture.h:166: UA_UNLOCK: Assertion `lock->mutexCounter == 1' failed.
```
## Background Information / Reproduction Steps

I have only tested with value callbacks but looking at [the code](https://github.com/open62541/open62541/blob/master/src/server/ua_services_attribute.c#L1490-L1515) I suspect the crash would happen with data sources and historizing as well.

Publisher is the `tutorial_pubsub_publish.c`. Subscriber is the `tutorial_pubsub_subscribe.c` with value callback added:

```diff
diff --git a/examples/pubsub/tutorial_pubsub_subscribe.c b/examples/pubsub/tutorial_pubsub_subscribe.c
index 881de73..066b8e1 100644
--- a/examples/pubsub/tutorial_pubsub_subscribe.c
+++ b/examples/pubsub/tutorial_pubsub_subscribe.c
@@ -133,12 +133,17 @@ addDataSetReader(UA_Server *server) {

     retval |= UA_Server_addDataSetReader(server, readerGroupIdentifier, &readerConfig,
                                          &readerIdentifier);
     return retval;
 }

+static void cb(UA_Server* server, const UA_NodeId* sessionId, void* sessionContext,
+               const UA_NodeId* nodeId, void* nodeContext, const UA_NumericRange* range,
+               const UA_DataValue* data)
+{}
+
 /**
  * **SubscribedDataSet**
  *
  * Set SubscribedDataSet type to TargetVariables data type.
  * Add subscribedvariables to the DataSetReader */
 static UA_StatusCode
@@ -190,12 +195,14 @@ addSubscribedVariables (UA_Server *server, UA_NodeId dataSetReaderId) {
         retval |= UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, (UA_UInt32)i + 50000),
                                            folderId,
                                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                                            UA_QUALIFIEDNAME(1, (char *)readerConfig.dataSetMetaData.fields[i].name.data),
                                            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                                            vAttr, NULL, &newNode);
+       UA_ValueCallback callback = {.onRead=cb, .onWrite=cb};
+       UA_Server_setVariableNode_valueCallback(server, newNode, callback);

         /* For creating Targetvariables */
         UA_FieldTargetDataType_init(&targetVars[i].targetVariable);
         targetVars[i].targetVariable.attributeId  = UA_ATTRIBUTEID_VALUE;
         targetVars[i].targetVariable.targetNodeId = newNode;
     }
```

Stack trace:

```diff
#0  0x00007ffff740bc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff740f028 in __GI_abort () at abort.c:89
#2  0x00007ffff7404bf6 in __assert_fail_base (fmt=0x7ffff7559058 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x50b71f "lock->mutexCounter == 1",
    file=file@entry=0x50b6d0 "/media/projects/open62541/arch/posix/ua_architecture.h", line=line@entry=166, function=function@entry=0x50bfd7 <__PRETTY_FUNCTION__.5987> "UA_UNLOCK") at assert.c:92
#3  0x00007ffff7404ca2 in __GI___assert_fail (assertion=0x50b71f "lock->mutexCounter == 1", file=0x50b6d0 "/media/projects/open62541/arch/posix/ua_architecture.h", line=166, function=0x50bfd7 <__PRETTY_FUNCTION__.5987> "UA_UNLOCK")
    at assert.c:101
#4  0x000000000043772d in UA_UNLOCK (lock=0x775ef8) at /media/projects/open62541/arch/posix/ua_architecture.h:166
#5  0x000000000043c0f0 in writeNodeValueAttribute (server=0x7756c0, session=0x775cd0, node=0x7cafb0, value=0x7fffffffdb90, indexRange=0x7fffffffdb80) at /media/projects/open62541/src/server/ua_services_attribute.c:1500
#6  0x000000000043cd81 in copyAttributeIntoNode (server=0x7756c0, session=0x775cd0, node=0x7cafb0, wvalue=0x7fffffffdb60) at /media/projects/open62541/src/server/ua_services_attribute.c:1743
#7  0x000000000041651e in UA_Server_editNode (server=0x7756c0, session=0x775cd0, nodeId=0x7fffffffdb60, callback=0x43c671 <copyAttributeIntoNode>, data=0x7fffffffdb60) at /media/projects/open62541/src/server/ua_server_utils.c:280
#8  0x000000000043d42e in Operation_Write (server=0x7756c0, session=0x775cd0, context=0x0, wv=0x7fffffffdb60, result=0x7fffffffdb24) at /media/projects/open62541/src/server/ua_services_attribute.c:1823
#9  0x0000000000421c78 in UA_DataSetReader_process (server=0x7756c0, rg=0x7c7c40, dsr=0x7c5b80, msg=0x7cc040) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1267
#10 0x0000000000421f35 in processMessageWithReader (server=0x7756c0, readerGroup=0x7c7c40, reader=0x7c5b80, msg=0x7fffffffddb0) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1373
#11 0x0000000000422091 in processNetworkMessage (server=0x7756c0, connection=0x7c06d0, msg=0x7fffffffddb0) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1405
#12 0x00000000004225e2 in decodeAndProcessNetworkMessage (server=0x7756c0, connection=0x7c06d0, encodingMimeType=UA_PUBSUB_ENCODING_UADP, buf=0x7fffffffded0) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1561
#13 0x00000000004229f2 in decodeAndProcessFun (channel=0x7c07f0, cbContext=0x7fffffffe070, buf=0x7fffffffdfa0) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1662
#14 0x00000000004d0ba2 in UA_PubSubChannelUDPMC_receive (channel=0x7c07f0, transportSettings=0x7c7ca0, receiveCallback=0x42297c <decodeAndProcessFun>, receiveCallbackContext=0x7fffffffe070, timeout=1000)
    at /media/projects/open62541/plugins/ua_pubsub_udp_multicast.c:773
#15 0x0000000000422b40 in receiveBufferedNetworkMessage (server=0x7756c0, readerGroup=0x7c7c40, connection=0x7c06d0) at /media/projects/open62541/src/pubsub/ua_pubsub_reader.c:1701
#16 0x0000000000425223 in UA_ReaderGroup_subscribeCallback (server=0x7756c0, readerGroup=0x7c7c40) at /media/projects/open62541/src/pubsub/ua_pubsub_readergroup.c:777
#17 0x00000000004df8ab in processEntryCallback (context=0x7fffffffe1b0, te=0x7c7840) at /media/projects/open62541/arch/common/ua_timer.c:214
#18 0x00000000004505cb in __ZIP_ITER (fieldoffset=0, cb=0x4df84b <processEntryCallback>, context=0x7fffffffe1b0, elm=0x7c7840) at /media/projects/open62541/deps/ziptree.c:222
#19 0x00000000004df1c4 in UA_TimerTree_ZIP_ITER (head=0x774528, cb=0x4df84b <processEntryCallback>, context=0x7fffffffe1b0) at /media/projects/open62541/arch/common/ua_timer.c:37
#20 0x00000000004dfa6d in UA_Timer_process (t=0x7744e0, nowMonotonic=8888432111621) at /media/projects/open62541/arch/common/ua_timer.c:269
#21 0x00000000004d1e63 in UA_EventLoopPOSIX_run (el=0x774440, timeout=200) at /media/projects/open62541/arch/eventloop_posix.c:248
#22 0x000000000040ccc8 in UA_Server_run_iterate (server=0x7756c0, waitInternal=true) at /media/projects/open62541/src/server/ua_server.c:949
#23 0x00000000004ccdd6 in UA_Server_runUntilInterrupt (server=0x7756c0) at /media/projects/open62541/plugins/ua_config_default.c:115
#24 0x0000000000402ea6 in run (transportProfile=0x7fffffffe370, networkAddressUrl=0x7fffffffe380) at /media/projects/open62541/examples/pubsub/tutorial_pubsub_subscribe.c:316
#25 0x0000000000403066 in main (argc=1, argv=0x7fffffffe488) at /media/projects/open62541/examples/pubsub/tutorial_pubsub_subscribe.c:354
```

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=on -DUA_ENABLE_PUBSUB=on -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=on -DUA_LOGLEVEL=100 -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=off -DUA_FORCE_WERROR=off ..
```
(inb4 I'm using UA_FORCE_WERROR=off because old compiler.)

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.4-816-g73a60b5-dirty
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 14.04
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-15:
The same answer as for the other issue:

The problem is related to use of multi-threading. PubSub on 1.3 is not thread-safe.
This was however added to the master branch and will be part of the next 1.4 release.
The change is too invasive for backporting to 1.3.

If you can use the master branch, I advise to do so.
Otherwise disable multi-threading in the SDK. And potentially use an application-level mutex to protect.

This PR adds a warning to the build system.
https://github.com/open62541/open62541/pull/5641

closed 2023-02-15 by jpfr

JuhaSointusalo, 2023-02-15:
Thanks and uuhh... sorry. The crashes are actually from master branch. I guess when you build from master the version number isn't very clear. My bad, I'll do better next time.