open62541/open62541 #5646: Using server callbacks defined in UA_ValueCallback, there is no way to inform client about possible anomalies

hobbitdev, 2023-02-17:
## Description
We have to focus on this scenarios:

In a server instance, we have to use the callbacks defined in `UA_ValueCallback` in order to "answare back" to a client.
In case of a reading request or writing request, if something went wrong in server logic how could I inform the client about it ?

For example a client wants to write a value, yet such value is out of admitted server range, I can skip to write the phisichal value, but I should inform the client with a `UA_STATUSCODE_BADOUTOFRANGE` status.

Another example could be a client wants to read a value, but due to sensor or plant status in that particualr moment the data is not available, how could the server inform the cleint with a `UA_STATUSCODE_BADDATAUNAVAILABLE` status?

## Background Information / Reproduction Steps

Used CMake options:
cmake -S'my source path' \
	-B'my build path' \
    -DOPENSSL_ROOT_DIR=$OPENSSLPATH \
	-DCMAKE_BUILD_TYPE=Release \
	-DBUILD_SHARED_LIBS=OFF \
	-DUA_NAMESPACE_ZERO=FULL \
	-DUA_ENABLE_SUBSCRIPTIONS=ON \
    -DUA_ENABLE_DISCOVERY=ON \
	-DUA_ENABLE_ENCRYPTION=OPENSSL \
	-DUA_BUILD_SELFSIGNED_CERTIFICATE=ON \
	-DUA_ENABLE_METHODCALLS=ON \
	-DUA_ENABLE_NODEMANAGEMENT=ON \
	-DUA_LOGLEVEL=300

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version v1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Muddasir-Shakil, 2023-03-07:
I agree with you that there is no way to sent StatusCodes based on value read. However, it seems like not too difficult to implement. First thing you have to do is to add return type status code to the read/write callbacks, then check the return status of callbacks in `readValueAttributeFromNode()` call in ua_services_attribute.c. 
Seems like a good starting issue @jpfr ?

hobbitdev, 2023-03-18:
Excuse me @Muddasir-Shakil, I'd like to know whether this issue has been taken in consideration or perhaps there is a way to work out it even in the latest version v1.3.5

Thank you, regards
Rossano