open62541/open62541 #5647: Porting OPC-UA on Embedded systems (PIC32)

KnowingBat, 2023-02-17:
## Description
Hello developers. This week I managed to develop a small server on a Windows machine just to try out this library.
For the purposes of another project, I want to port this server to a PIC32 board that my colleagues created a few years ago. 
We already have an application running on it, which is a bare metal application (developed with Microchip Harmony v2.0) capable of interfacing an industrial machine and a web server. We now want to implement OPC-UA and we think that this library may be the right choice. Note that we do not have any kind of operating system on this board (e.g. freeRTOS, Azure RTOS).
Do you think it is possible to use this library with a bare metal application (so, without any OS)? If so, what are the main steps to do this?

## Checklist

 - [X] open62541 Version (release number or git tag): v1.3
 - [X] Other OPC UA SDKs used (client or server): None
 - [X] Operating system: Bare metal embedded firmware on PIC32MZ
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-02-19:
You should be able to port open62541.
For the 1.3 branch there is a freeRTOS port that uses the lwIP networking library.
Use that as the starting point.

Note that we are cleaning the the arch-support interfaces on master.
Things will change slightly for the upcoming releases.
But overall, we are keeping the architecture-support in the /arch folder with only well-defined interfaces to the main library.

KnowingBat, 2023-03-16:
Hello there, i just want to update this "issue". I have been working for a few weeks now on porting open62541 on PIC32.

KnowingBat, 2023-03-20:
Thank you for the information @jpfr.
I'm closing this issue because i found a real issue while porting on PIC32, described in #5698

closed 2023-03-20 by KnowingBat