open62541/open62541 #5649: UA_Client_run_iterate crashes in v 1.3.4

mario-reif, 2023-02-18:
Hi,

I have switched from open62541 v1.2.6 to v1.3.4 recently. After that my clients crashed when the OPCUA-Server was switched off (which is quite normal for industrial machines, which are taken from power at weekends for example).

To be sure that it is not a problem with my own code I wrote a very minimalistic server and client. Sourcecode for these is attached. My problem is very easy to reproduce: start server in one console, start client in another console, wait until client is connected, quit the server with CTRL-C. When I compile the server and the client once with open62541 v1.3.4 and once with v.1.2.6 I get different behaviours:
v1.2.6: after pressing CTRL-C to quit the server, the client recognizes that the server is not available any more and stops regularly
v1.3.4: after pressing CTRL-C to quit the server, the client crashes (process quits immediately) when the function UA_Client_run_iterate gets called.

**With UA_LOGLEVEL=200 I get the following logs with v1.3.4 before the process dies:**
[2023-02-18 12:27:27.100 (UTC+0100)] debug/channel	Connection 3 | SecureChannel 1 | Sending request with RequestId 11 of type ReadRequest
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
------------after UA_Client_run_iterate
------------before UA_Client_run_iterate
[2023-02-18 12:27:27.100 (UTC+0100)] debug/client	Async read response for request 11
------------after UA_Client_run_iterate
the content is: 42
[2023-02-18 12:27:28.100 (UTC+0100)] debug/channel	Connection 3 | SecureChannel 1 | Sending request with RequestId 12 of type ReadRequest
------------before UA_Client_run_iterate
[2023-02-18 12:27:28.101 (UTC+0100)] warn/channel	Connection 3 | SecureChannel 1 | Receiving the response failed with StatusCode BadConnectionClosed
[2023-02-18 12:27:28.101 (UTC+0100)] debug/client	Async read response for request 12
./execute_client: Zeile 3: 44668 Speicherzugriffsfehler  LD_LIBRARY_PATH=$LD_LIBRARY_PATH:../external_lib/install/lib/ ./client

**CMake Options:**
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Release
CMAKE_INSTALL_PREFIX:PATH=/install
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_FORCE_WERROR:BOOL=ON
UA_LOGLEVEL:STRING=600
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=Full

**Thats how I build the shared lib**
cmake -DOPEN62541_VERSION=v1.3.4 -DCMAKE_INSTALL_PREFIX=../install -DCMAKE_BUILD_TYPE=Release -DUA_NAMESPACE_ZERO=Full -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=200 $SRC
make clean
make -j
make install

 - [ ] open62541 Version (release number or git tag): v1.3.4
 - [ ] Other OPC UA SDKs used (client or server): none
 - [ ] Operating system: Debian 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached: yes
 - [ ] Critical issue

Best Regards,
Mario
[client.zip](https://github.com/open62541/open62541/files/10773974/client.zip)
[server.zip](https://github.com/open62541/open62541/files/10773975/server.zip)

jpfr, 2023-02-19:
The fix is in https://github.com/open62541/open62541/pull/5650.
The situation was also reproduced in a unit test to prevent future regressions.

Please verify that this solves the issue on your end.

mario-reif, 2023-02-19:
No, it didn't work for me.

I used the sources from Release 1.3.4 and changed the function UA_Client_clear in the file ua_client.c as you described in your pull request #5650 . I also patched UA_Client_Subscriptions_backgroundPublish, but I don't use subscriptions, so I think this is not important. I also added a Debug-Log-Message to see if the code crashes at this function. My UA_Client_clear function now looks like this:

```
UA_Client_clear(UA_Client *client) {
    UA_LOG_DEBUG(&UA_Client_getConfig(client)->logger, UA_LOGCATEGORY_CLIENT, "UA_Client_clear entered.");

    /* Prevent new async service calls in UA_Client_AsyncService_removeAll */
    UA_SessionState oldState = client->sessionState;
    client->sessionState = UA_SESSIONSTATE_CLOSING;

    /* Delete the async service calls with BADHSUTDOWN */
    UA_Client_AsyncService_removeAll(client, UA_STATUSCODE_BADSHUTDOWN);

    /* Reset to the old state to properly close the session */
    client->sessionState = oldState;

    UA_Client_disconnect(client);
    UA_String_clear(&client->endpointUrl);

    UA_String_clear(&client->remoteNonce);
    UA_String_clear(&client->localNonce);

    /* Delete the subscriptions */
#ifdef UA_ENABLE_SUBSCRIPTIONS
    UA_Client_Subscriptions_clean(client);
#endif

    /* Delete the timed work */
    UA_Timer_clear(&client->timer);
}
```
I recompiled the open62541 library, my test-server and my test-client. Same result like yesterday. It seems to me, that the program crashes before the function UA_Client_clear gets called, because in the logs my new log-message in the first line of UA_Client_clear is not visible:

```
[2023-02-19 15:32:03.699 (UTC+0100)] debug/client	Async read response for request 9
------------after UA_Client_run_iterate
the content is: 42
[2023-02-19 15:32:04.699 (UTC+0100)] debug/channel	Connection 3 | SecureChannel 1 | Sending request with RequestId 10 of type ReadRequest
[2023-02-19 15:32:04.699 (UTC+0100)] trace/channel	Connection 3 | SecureChannel 1 | Send from a symmetric message buffer of length 131072 a message of header+payload length of 121
[2023-02-19 15:32:04.699 (UTC+0100)] trace/channel	Connection 3 | SecureChannel 1 | Send from a symmetric message buffer of length 131072 a message of length 121
------------before UA_Client_run_iterate
[2023-02-19 15:32:04.699 (UTC+0100)] warn/channel	Connection 3 | SecureChannel 1 | Receiving the response failed with StatusCode BadConnectionClosed
[2023-02-19 15:32:04.699 (UTC+0100)] debug/client	Async read response for request 10
./execute_client: Zeile 3: 35561 Speicherzugriffsfehler  LD_LIBRARY_PATH=$LD_LIBRARY_PATH:../external_lib/install/lib/ ./client

```

mario-reif, 2023-02-20:
I tried to find the code in open62541 more precisely, where the library crashes. With some more log-messages added to my open62541 local code, I can provide this call-stack of functions before the crash:

My programm calls the function UA_Client_run_iterate, this is where my call stack starts. In the line under the function the code line is shown, where the function calls another function

```
ua_client.c             function UA_Client_run_iterate
                        last call in this function: retval = receiveResponse(client, NULL, NULL, maxDate, NULL);
ua_client.c             function receiveResponse
                        last call in this function: closeSecureChannel(client);
ua_client_connect.c     function closeSecureChannel
                        last call in this function: UA_Client_AsyncService_removeAll(client, UA_STATUSCODE_BADSECURECHANNELCLOSED);
ua_client.c             function UA_Client_AsyncService_removeAll
                        last call in this function: UA_Client_AsyncService_cancel(client, ac, statusCode);
ua_client.c             function UA_Client_AsyncService_cancel
                        last call in this function: ac->callback(client, ac->userdata, ac->requestId, &response);
```

At ac->callback I cannot investigate more, because I don't know which function gets called by the callback.

The library crashes at ac->callback. I hope this helps.

jpfr, 2023-02-20:
The PR actually did fix.
Now the problem is different.

The API says that the value is NULL when the async service call fails.
Just check the status code before handling the value in the callback.
https://github.com/open62541/open62541/blob/7c3a971b71161d6be9d019138726fd7bc2f603e9/include/open62541/client_highlevel_async.h#L58

closed 2023-02-20 by jpfr