open62541/open62541 #5664: open62541 doesn't compile with EXAMPLES and PUBSUB together

danmoser, 2023-03-02:
## Description
open62541 doesn't compile with `UA_BUILD_EXAMPLES` and `UA_ENABLE_PUBSUB` options together. 

## Background Information / Reproduction Steps
I can build open62541 if `UA_BUILD_EXAMPLES=ON` OR `UA_ENABLE_PUBSUB=ON`, but not with both options simultaneously enabled.

Trying to compile this:

`cmake .. -DBUILD_SHARED_LIBS=ON -DCMAKE_C_COMPILER=/usr/bin/gcc-9 -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON`

Returns the following:
```
[ 89%] Built target tutorial_pubsub_publish
[ 89%] Linking C executable ../bin/examples/tutorial_pubsub_publish_raw
[ 89%] Building C object examples/CMakeFiles/server_pubsub_publish_rt_level.dir/pubsub/server_pubsub_publisher_rt_level.c.o
[ 89%] Linking C executable ../bin/examples/server_pubsub_publisher_iop
[ 89%] Linking C executable ../bin/examples/tutorial_pubsub_subscribe_raw
[ 91%] Built target tutorial_pubsub_publish_raw
[ 91%] Linking C executable ../bin/examples/server_pubsub_publish_rt_level
[ 91%] Building C object examples/CMakeFiles/server_pubsub_publish_rt_level_raw.dir/pubsub/server_pubsub_publisher_rt_level_raw.c.o
/usr/bin/ld: /tmp/tutorial_pubsub_subscribe_raw.BRDw2e.ltrans0.ltrans.o: in function `addPubSubConnection':
/data/open62541_v1.3/examples/pubsub/tutorial_pubsub_subscribe_raw.c:70: undefined reference to `UA_PubSubConnection_regist'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/tutorial_pubsub_subscribe_raw.dir/build.make:98: bin/examples/tutorial_pubsub_subscribe_raw] Error 1
make[1]: *** [CMakeFiles/Makefile2:1598: examples/CMakeFiles/tutorial_pubsub_subscribe_raw.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Built target server_pubsub_publisher_iop
[ 91%] Built target server_pubsub_publish_rt_level
[ 92%] Linking C executable ../bin/examples/server_pubsub_publish_rt_level_raw
[ 92%] Built target server_pubsub_publish_rt_level_raw
make: *** [Makefile:156: all] Error 2
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.3
 - [ ] cmake Version: 3.22.1 
 - [ ] Operating system: Ubuntu 22.04

danmoser, 2023-03-03:
Just tested with gcc11. Same error

JohannesKauffmann, 2023-05-25:
@jpfr This issue can be closed.

closed 2023-06-07 by NoelGraf