open62541/open62541 #5667: Generated NodeIds for Methods same as for type instances

robertns, 2023-03-06:
Issue detected in 1.3 (also in earlier versions, I tried with 1.1 and 1.0).

Generated nodeIds for (mandatory) methods are the same as for type instances for those methods. This results in having multiple parents for that nodeId, theoretically typeinstance nodeid can have unlimited number of parents.

Simple example which shows this can be made with 1.0 version of the library, with UA_NAMESPACE_ZERO set to FULL.

TemporaryFileTransferType is suitable for reproducing, since it has mandatory methods as components.

When the following lines are inserted to example server code
```c
/* 3) Add the node */
    UA_Server_addVariableNode(server, newNodeId, parentNodeId, parentReferenceNodeId,
                              browseName, variableType, attr, NULL, NULL);

  /* Begin test code */
   UA_NodeId newTFTNodeId = UA_NODEID_STRING(1, "temporaryFileTransfer");
   UA_NodeId tftTypeDef = UA_NODEID_NUMERIC(0, UA_NS0ID_TEMPORARYFILETRANSFERTYPE);
   UA_QualifiedName browseTFTName1 = UA_QUALIFIEDNAME(1, "tft1");
   UA_ObjectAttributes objAttr = UA_ObjectAttributes_default;
   UA_Server_addObjectNode(server, newTFTNodeId, parentNodeId, parentReferenceNodeId, browseTFTName1, tftTypeDef, objAttr, 0, 0);

   UA_QualifiedName browseTFTName2 = UA_QUALIFIEDNAME(1, "tft2");
   UA_NodeId newTFTNodeId2 = UA_NODEID_STRING(1, "temporaryFileTransfer2");
   UA_Server_addObjectNode(server, newTFTNodeId2, parentNodeId, parentReferenceNodeId, browseTFTName2, tftTypeDef, objAttr, 0, 0);
  /* End test code */
   
    /* Run the server loop */
    UA_StatusCode status = UA_Server_run(server, &running);
```

As a result, two new objects tft1 and tft2 are constructed, both with methods CloseAndCommit (i=15751, which is TemporaryFileTransferType_CloseAndCommit), GenerateFileForRead (i=15746), and GenerateFileForWrite (i=15749).

As a result, in Types tree, method components of TemporaryFileTransferType (i=15744) each have three "ComponentOf" references, to TemporaryFileTransferType, tft1 and tft2.

This behavior doesn't occur with properties, for them the new nodeId is generated.

ccvca, 2023-03-06:
This is intended. These NodeIds are not duplicated. The nodes are used in multiple instances. To distinguish between the different instances, each node has a context poitner to the containing object in the callback method.

robertns, 2023-03-14:
Thank you for clarification!

closed 2023-03-14 by robertns