open62541/open62541 #5675: Dangling logger pointers on preset SecurityPolicies

MattesWhite, 2023-03-09:
## Description

When we create a `UA_ServerConfig` and add security policies before moving the config on `UA_Server_newWithConfig()` the logger pointers in the security policy context become invalidated as they still point to the original stack-allocated `UA_ServerConfig` instance.

## Background Information / Reproduction Steps

1. Create a `UA_ServerConfig` variable on the stack
1. Initialize the config with cryptographic security policies
   - e.g. create a self-signed certificate und use `UA_ServerConfig_setDefaultWithSecurityPolicies()`
1. Move the config into a new server via `UA_Server_newWithConfig`
2. The pointer to logger in the respective `UA_SecurityPolicy->context` become invalid as they still point to the stack variable
3. In my case this lead to segfaults in the server when disconnecting clients using `Sign` or `SignAndEncrypt` modes.

Used CMake options:

Nothing special except OpenSSL encryption

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.3
 - [x] Other OPC UA SDKs used (client or server): server side but it may also happen at the client side as they use the same code for secure channels
 - [x] Operating system: Windows 10 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2023-03-13:
Note: Same problem with PKI plugins has been resolved with PR #5583 in master branch by moving logger from unknown context to structure and updating pointers in UA_Server_newWithConfig.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier