open62541/open62541 #5676: Two different servers on the same machine but using the same IP address with different TCP port number

hobbitdev, 2023-03-10:
## Description
Perhaps I've misundertood the meaning of the attribute `UA_ServerConfig.customHostname`, but I'm not able to configure two different OPCUA server using the same IP address with different TCP port number.

I need to instance two different server on the same IP address like this:
Server 1) 192.168.85.40:4840
Server 2) 192.168.85.40:4845
The two server contents are different, I mean they publish different variables.

Below the code I'm using in order to achive it, but it doesn't behave as expected, as I get only the second server runing but with its TCP port on 4840 and not 4845 as it should be.
The first server is not reachable at all.
Of course the code below is parameterized, it works perfectly with an instance only.
```
// Set the IP address if given
if (pAddressIP) pServer->pSc->customHostname = UA_STRING_ALLOC(pAddressIP);

// Creates a new server config with one endpoint and custom buffer size
status = UA_ServerConfig_setMinimalCustomBuffer(pServer->pSc, tcpport, NULL, sendBufferSize, recvBufferSize);
```
What I'm doing wrong?
Could someone point me to the right direction in order to configure two different server on the same machine ?

## Background Information / Reproduction Steps

Used CMake options:
cmake -S'my source path' \
	-B'my build path' \
    -DOPENSSL_ROOT_DIR=$OPENSSLPATH \
	-DCMAKE_BUILD_TYPE=Release \
	-DBUILD_SHARED_LIBS=OFF \
	-DUA_NAMESPACE_ZERO=FULL \
	-DUA_ENABLE_SUBSCRIPTIONS=ON \
    -DUA_ENABLE_DISCOVERY=ON \
	-DUA_ENABLE_ENCRYPTION=OPENSSL \
	-DUA_BUILD_SELFSIGNED_CERTIFICATE=ON \
	-DUA_ENABLE_METHODCALLS=ON \
	-DUA_ENABLE_NODEMANAGEMENT=ON \
	-DUA_LOGLEVEL=300

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version v1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-03-13:
On master we parse the server url and extract the hostname and port number.
But your are on the 1.3 branch.

There the port number is baked into the network layer.
When you create the second server, the network layer is initialized via UA_ServerConfig_addNetworkLayerTCP.
Probably you create a new server with the default settings and only change the port number afterwards.
Have a look at the default server configuration plug-in.

hobbitdev, 2023-03-18:
Excuse me @jpfr, it's my fault, but I'm not able to understand your tip ...

Now, I'm using the version **v1.3.5**, is this the problem? Should I switch to **master** branch ?
May I instance more than one server with the version 1.3.5 ?

Below is the code I'm using to configure my server instances, before starting the single server instance using `UA_Server_run`
I simply call `UA_ServerConfig_setMinimalCustomBuffer` after the server pointer creation with `UA_Server_new`
After `UA_Server_new() `there are no networks layers defined (see picture 1), it "appears" after `UA_ServerConfig_setMinimalCustomBuffer() ` (see picture 2)
The only things I'm able to notice is the discovery URL is NULL, is it right or wrong?

Based on the code I'm using, I shouldn't create the server using default settings ... for sure I'm wrong, but I don't see it.

```
// Create a new server with default plugins for logging etc. used during initialization. 
// No network layer and SecurityPolicies are set so far.
pServer = UA_Server_new();

// Set the IP address if given
if (pAddressIP) UA_Server_getConfig(pServer)->customHostname = UA_STRING_ALLOC(pAddressIP);

// Creates a new server config with one endpointand custom buffer size
status = UA_ServerConfig_setMinimalCustomBuffer(UA_Server_getConfig(pServer), tcpport, NULL, sendBufferSize, recvBufferSize);
```

**picture 1**
[image]

**picture 2**
[image]

Regards
Rossano

hobbitdev, 2023-03-24:
Excuse me @jpfr, I have to say that this issue has nothing to do with the library.
It works perfectly even with different server instances.

The issue was due to my poor library knoledge.
Server instances were correctly created, but at the moment of running them I was using a single thread instead of two.

Best regards
Rossano