open62541/open62541 #5677: open62541 with KEPServerEx: cannot access with user credentials

eschmidt-WI, 2023-03-10:
## Description

When you try to connect to the open62541 opc-ua-server (/examples/server_ctt, version 1.3.5) via user credentials with the [KEPServerEx ](https://www.ptc.com/de/products/kepware/kepserverex) tool it gives the error message: `Endpoint not supported.` Without user credentials the connection works well.

I don't know if this is an bug at the KEPServerEx or at the new open62541 Stack version. Version 1.0.2 of open62641 works well with the KEPServerEx. Also other clients works well.

## Background Information / Reproduction Steps

1. Clone and checkout open62541 to branch `v1.3.5`.

2. Build the examples inside a docker container: (I had an issue with buildx so I removed the second image stage inside the Dockerfile)
`docker build -f tools/docker/Dockerfile -t open62541:local-build`

3. Start the container and run a shell inside it:
`docker run -it --network host open62541:local-build sh`

3. Create a certificate via openssl **e. g**.: (First create localhost.cnf then the certificate)
```
/opt/open62541 # openssl req \
             -new \
             -nodes \
             -x509 \
             -sha256 \
             -newkey rsa:2048 \
             -keyout localhost.key \
             -days 3650 \
             -subj "/C=DE/O=YOURORGANIZATION/CN=REPLACETHIS" \
             -out localhost.crt -config localhost.cnf
/opt/open62541 # openssl x509 \
             -in localhost.crt \
             -outform der \
             -out cert.der
/opt/open62541 # openssl rsa \
             -info -inform PEM \
             -in localhost.key \
             -outform DER \
             -out key.der
```
4. Start the example server:
`./build/bin/examples/server_ctt cert.der key.der`
5. Create a channel and device in KEPServerEx (Use 127.0.0.1:4840 and user1, password)
6. Open the configuration of the device and start `tag-generation` (Also other connection methods throws these error). Now the error happend and the KEPServerEx prints:
```
Kanal1 | Kanal konnte nicht verbunden werden. | Status = 'Der Endpunkt wird vom Server nicht unterstützt.'
```
The example server prints:
```
/opt/open62541 # ./build/bin/examples/server_ctt cert.der key.der
[2023-03-10 15:34:08.140 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-03-10 15:34:08.140 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
[2023-03-10 15:34:08.167 (UTC+0000)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=integer): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-03-10 15:34:08.167 (UTC+0000)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=uinteger): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-03-10 15:34:08.172 (UTC+0000)] info/network       TCP network layer listening on opc.tcp://DE13680:4840/
[2023-03-10 15:34:10.158 (UTC+0000)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2023-03-10 15:34:10.158 (UTC+0000)] info/channel       Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-03-10 15:34:10.159 (UTC+0000)] info/channel       Connection 5 | SecureChannel 1 | CloseSecureChannel
[2023-03-10 15:34:10.159 (UTC+0000)] info/network       Connection 5 | Closed
```
Same issue with Basic256Sha256:
```
[2023-03-10 15:40:08.458 (UTC+0000)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2023-03-10 15:40:08.465 (UTC+0000)] warn/server        PKI plugin unconfigured. Accepting the certificate.
[2023-03-10 15:40:08.471 (UTC+0000)] info/channel       Connection 5 | SecureChannel 4 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 300.00s
[2023-03-10 15:40:08.486 (UTC+0000)] info/session       SecureChannel 4 | Session "Kepware_KEPServerEX_UA_Client_Driver:Kanal2" | Session created
[2023-03-10 15:40:08.492 (UTC+0000)] info/session       SecureChannel 4 | Session "Kepware_KEPServerEX_UA_Client_Driver:Kanal2" | Closing the Session
[2023-03-10 15:40:08.492 (UTC+0000)] info/channel       Connection 5 | SecureChannel 4 | CloseSecureChannel
[2023-03-10 15:40:08.492 (UTC+0000)] info/network       Connection 5 | Closed
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.5 (1.3.3 also)
 - [x] Other OPC UA SDKs used (client or server):  /examples/server_ctt
 - [x] Operating system: Ubuntu
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
[KEPServerEx_open62541_user_credentials_bug.zip](https://github.com/open62541/open62541/files/10943552/KEPServerEx_open62541_user_credentials_bug.zip)

 - [x] Self-contained code example attached: server_ctt example from this repository
 - [x] Critical issue: Can't login, only without credentials (insecure)

jpfr, 2023-03-13:
I was able to reproduce.
The KEPWare client opens a SecureChannel.
It then opens a Session and aborts.

[image]

What is missing is the "ActiveSessionRequest" to be sent by the KEPWare client.

The log messages on the KEPWare side are not very helpful.
Do you know a way of further increasing the log level?

eschmidt-WI, 2023-03-13:
No, I don't know a way to increasing the log level on the KEPWare side. 

I compared the endpoints from the GetEndpointsResponse the only thing I noticed is that the DiscoveryUrl is present in 1.3.5. In previous versions this was not present.

jpfr, 2023-03-15:
I tried to connect kepware to some other servers. Like the prosys simulation server. Doesn’t work either.

we could try what happens for different/older security policies.
The problem seems to originate on their side though…

o6-nbeier, 2026-10-20:
Closing for no activity.

closed 2026-10-20 by o6-nbeier