open62541/open62541 #5685: How many threads does the server have in MULTITHREADING mode?

vvpnet, 2023-03-15:
Hi guys. Tell me how many threads can there be in the server?

I have the following problem. Clients request historical data (it takes some time) at this point other clients cannot request historical data on other nodes. How can these requests be separated?

xydan83, 2023-03-20:
Hello. As far as I know the server has one worker thread but it has an event loop. Multithreading mode (https://www.open62541.org/doc/master/building.html#build-options) is needed for thread-safe external calls. For example, if you want to call some functions from a pair of threads you created to eliminate the race for resources. Essentially, such server or client API functions are marked with the UA_THREADSAFE macro.

Well, I think your historical request from one client is very thread-loading on the server and it can't execute a new request until it completes the previous one.

> UA_MULTITHREADING
> Level of multi-threading support. The supported levels are currently as follows:
> 
> 0-99: Multithreading support disabled.
> 
> '>=100: API functions marked with the UA_THREADSAFE-macro are protected internally with mutexes. Multiple threads are allowed to call these functions of the SDK at the same time without causing race conditions. Furthermore, this level support the handling of asynchronous method calls from external worker threads.

vvpnet, 2023-03-20:
Proceeding from what has been said. I need to make it so that getting historical data takes little time. There don't seem to be any other options.

xydan83, 2023-03-20:
> Proceeding from what has been said. I need to make it so that getting historical data takes little time. There don't seem to be any other options.

I think so. I think you need to query the data in small chunks. If it's real for you...

o6-nbeier, 2026-10-20:
Closing for no activity.

closed 2026-10-20 by o6-nbeier